riscv/riscv-debug-spec #586: mcontrol match=1 definition when maskmax=0

JamesKenneyImperas, 2020-11-05:
Something doesn't seem right about the definition of `mcontrol` when `match=1` and `maskmax=0`. Firstly, the specification states for the `maskmax` field:

> A value of 0 indicates that only exact value matches are supported (one byte range).

Then, the definition of `match=1` behavior for `mcontrol` is:

> 1: Matches when the top M bits of the value match the top M bits of tdata2. M is XLEN-1 minus the index of the least-significant bit containing 0 in tdata2. Debuggers should only write values to tdata2 such that M + maskmax ≥ XLEN, otherwise it’s undefined on what conditions the trigger will fire.

When `maskmax=0`, there is no value that can be written to `tdata2` for which the condition `M + maskmax ≥ XLEN` can be true. Either the definition of the `maskmax` field is incorrect, or the definition of the `match=1` case needs correcting for the case when `maskmax=0`. Perhaps the wording when `match=1` needs to change to:

> 1: If maskmax=0, equivalent to match=0. If maskmax!=0, matches when the top M bits of the value match the top M bits of tdata2. M is XLEN-1 minus the index of the least-significant bit containing 0 in tdata2. Debuggers should only write values to tdata2 such that M + maskmax ≥ XLEN, otherwise it’s undefined on what conditions the trigger will fire.

(Incidentally, the word _Debuggers_ should be removed from this text, since the trigger module is accessible to M-mode code too.)

The text for `match=1` should also specify the behavior when all-ones is written to `tdata2`; is this the same as for mcontrol6, or something else?

scottj97, 2020-11-05:
I assume the intent was that if match=1 is supported, maskmax would be >0. Otherwise it doesn't really make sense to support match=1 since it would be the same as match=0.

I agree it could be clarified, and your suggestion makes sense.

timsifive, 2020-12-17:
Changing:
1. maskmax=0 means match=1 is unsupported.
2. Writing all ones to tdata2 with match=1 means it's undefined when the trigger fires.

> (Incidentally, the word Debuggers should be removed from this text, since the trigger module is accessible to M-mode code too.)

Debuggers aren't limited to things running remotely over JTAG. M-mode software that sets triggers is likely also a debugger. (There probably are some non-debug uses for triggers, but I don't think it's worth mentioning them because it'll make the language more complex.)

closed 2020-12-19 by timsifive