riscv/riscv-debug-spec #617: How to access VPR?

tsaiyenting, 2021-02-18:
Hi all,

May I have some question about using abstract command to access VPR(VLEN=256) and FPR(FLEN=64) if DXLEN=32?
The riscv-debug-spec seems not define regno for VPR yet. Will there be more than 1 index for single VPR since COMMAND.aarsize(at most 4 for 128-bit) is smaller than 256-bit?

I am also confused about accessing FPR(FLEN=64) when DXLEN=32.
For example, if I want to write FPR0, debugger should
1. write {DATA1, DATA0} for new FPR0 even DXLEN=32
2. write COMMAND.aarsize=0x3, COMMAND.regno=0x1020, COMMAND.write=0x1
Is it correct?

Thanks

pdonahue-ventana, 2021-02-20:
> May I have some questions about using abstract command to access VPR(VLEN=256) and FPR(FLEN=64) if DXLEN=32?
> The riscv-debug-spec seems not define regno for VPR yet. Will there be more than 1 index for single VPR since COMMAND.aarsize(at most 4 for 128-bit) is smaller than 256-bit?

Yes, the V extension is not yet supported by abstract commands.  I don't know if it makes more sense to define aarsize values for wider registers or to have VPRs consume more than one regno, though my intuition is that the former is better.

> I am also confused about accessing FPR(FLEN=64) when DXLEN=32.
> For example, if I want to write FPR0, debugger should
> 
> 1. write {DATA1, DATA0} for new FPR0 even DXLEN=32
> 2. write COMMAND.aarsize=0x3, COMMAND.regno=0x1020, COMMAND.write=0x1
>    Is it correct?

You also need transfer=1.  Table 3.1 shows that if the argument is 64b then the value is in data0 and data1 with the least significant word in data0 so step 1 is correct.  Assuming transfer=1, it looks like your values in step 2 correctly indicate a write of size 64b to f0.

timsifive, 2021-02-20:
The current V spec draft doesn't seem to put any requirements on VLEN except that it is a power of 2. So the debug spec should be prepared to handle VLEN of at least 2**16, and ideally larger. I think that rules out using additional aarsize values, and makes the whole thing of assigning numbers pretty messy. Assuming aarsize=128 bits, we'd need 9 bits per vector register, and there are 32 of them, so 14 bits total. Current abstract access register command reserves "just" 16 bits for the register number, so that's pretty prohibitive. (There's only 1 unused bit in abstract access register command.)

Debug implementations that have a program buffer don't need to worry about this at all. The relevant instructions exist and the debugger can simply use them. (OpenOCD does this.)

We could add abstract vector command that mirror the instructions that a debugger would place in the program buffer, but that feels like double work. I'm inclined to say that people should just implement a program buffer if they want to access vector registers. When this debate was happened, the abstract-command camp primarily pointed to hardware size as a reason not to implement the program buffer approach. If you have a vector unit then you can spare the gates for a program buffer debug implementation.

Thoughts?

tsaiyenting, 2021-02-22:
Yes. I agreed it is not realistic to add abstract vector command. 
Program buffer method can handle this situation and keep hardware design simple.

Thanks

closed 2021-02-23 by tsaiyenting