riscv/riscv-debug-spec #619: Design constrained halt group?

stanleyleee, 2021-02-26:
Hi all,
    In specification of 1.0.0 stable, it mentions that hardcoded halt group is explicitly allowed. However, I am wondering if the following scenario is also allowed.

- Hart 0,1 are pre-defined in halt group 1 and are required always in the same halt group

- Hart 2, 3 are pre-defined in halt group 2 and are required always in the same halt group

- Debugger writes dmcs2 to move hart 2 to halt group 1, but design also moves hart 3 to halt group 1 (which is not the intention of debugger).

Thanks

pdonahue-ventana, 2021-02-26:
Hard-coded values aren't writable.  If harts 2 and 3 are hard-coded into halt group 2 then the debugger cannot move hart 2 into halt group 1 (or any other halt group) and the design cannot move hart 3 into halt group 1 (or any other halt group).

stanleyleee, 2021-03-02:
Hi Donahue,
   
Thanks for your response that hardwire halt groups can not be modified.

It seems the case above should not be configured as hardwire halt groups.

If an implementation expect HARTs in the same CORE should enter halt state together because of shared resources,
Are there any suggested ways to provide the flexibility to configure COREs into same or different halt groups.

Many Thanks

pdonahue-ventana, 2021-03-04:
In your example implementation with two cores that each contain two harts, I think that you can certainly hardcode harts 0 and 1 to halt group 1 and harts 2 and 3 to halt group 2.  Or you could hardcode all four harts to group 1 but then you can't independently halt the independent cores.  Either way, things are hardcoded which eliminates the flexibility to decide whether you want to hart cores independently or together.

It seems like a reasonable use case to want the ability to dynamically put the two cores in separate groups or the same group.  All harts would start up in group 1 (since they can't be in group 0 and group 1 is the lowest group number they can be in).  You could put hart 0 in group 2 but then I think that in your implementation hart 1 ends up in group 2.  I don't think that the non-explicit change of the group for hart 1 is allowed by the spec unless there's something preventing you from selecting only hart 0 to start with.  @timsifive, do you have any ideas?

stanleyleee, 2021-03-16:
Hi,

Here is another question about halt, resume group and external trigger.

There is a description of dmcs2, "The DM external triggers available to add to halt groups may be the same as or distinct from the DM external triggers available to add to resume groups".

Which of the following cases satisfy the description?

1: An external trigger can be added to a halt group and a resume group at the same time.

2: An external trigger can be added to either a halt group or a resume group, but not in a halt group and a resume group at the same time.

Thanks

pdonahue-ventana, 2021-03-17:
There are some external triggers which can cause a group of harts to halt.  There are some (possibly same or possibly different) external triggers which can cause a group of harts to resume.  Harts can be added and removed from halt/resume groups independently (unless they're hard-wired).

1. I'm not sure exactly what "can be added ... at the same time" means.  If it means that you program halt group H to halt on trigger N and then you separately program a resume group R to resume on trigger N then that's certainly fine.  If you mean that you program halt group H to halt on trigger N and then that has the side-effect of programming resume group R to resume on trigger N then my interpretation is that this type of side-effect is not allowed.  It seems like a debugger must be assured that when it sets things up in a particular way that there are no side-effects that are not explicitly allowed so once R is configured then configuring H shouldn't affect R.
2. This seems like once halt group H is programmed to halt on trigger N then trigger N disappears from the resume group.  I also don't think that's allowed.  I don't see any explicit prohibition of that but it seems like once a debugger does discovery and finds that trigger N is available for halt groups and that trigger N is available for resume groups that it must be able to assume that this won't change dynamically as a side-effect of some other configuration.

stanleyleee, 2021-03-19:
Thanks a lot.  It seems hardware should always follow the setting of debugger without side-effect.

There are some scenarios which confuse me.

For example,

     1. Trigger N is belong to halt group H and resume group R.  
         How should we distinguish that Trigger-Out of N is representing event from halt group H or resume group R. 

     2. Hart 0 and Trigger N are both belong to halt group H and resume group R.
         When Trigger-In of N is set, it seems hart 0 should be halted since halt take priority over resume.

By the way, I think selecting multiple harts with hawindow and hasel can configure multiple hart with single dmcs2 command with hgwrite =1, hgselect =0, because it mentions "all selected harts" in description of hgwrite.
However, in "3.3.2 Selecting Multiple Harts", there is a statement that "Only the actions initiated by dmcontrol can apply to multiple harts at once".   Is dmcs2 also honor to the setting of hawinodw and hasel?

Many Thanks

pdonahue-ventana, 2021-03-22:
>  1. Trigger N is belong to halt group H and resume group R.  
     How should we distinguish that Trigger-Out of N is representing event from halt group H or resume group R. 

I think you're saying that output trigger N is notified when the harts in group H halt and it is also notified when the harts in group R resume.  I don't think that you can tell which of the two situations caused N to be notified.  If you need to do that then it seems like you should use two different triggers.

>  2. Hart 0 and Trigger N are both belong to halt group H and resume group R.
     When Trigger-In of N is set, it seems hart 0 should be halted since halt take priority over resume.

If a debugger programmed trigger N to tell hart 0 both to halt and to resume then that seems like it's probably a bug in the debugger.  But regardless of real-world usefulness, it's an interesting question.  I don't remember whether the spec specifies the priority of haltreq over resumereq in dmcontrol.  If that rule exists then it seems like it should also apply here.  If the rule doesn't exist then it should probably be explicitly specified or described as UNSPECIFIED.

pdonahue-ventana, 2022-05-27:
This was addressed in #642.

closed 2022-05-27 by pdonahue-ventana