riscv/riscv-debug-spec #638: Clarification of dcsr.stopcount and dcsr.stoptime bits

stnolting, 2021-06-04:
While implementing the RISC-V debug mode I came across the `dcsr.stopcount` and `dcsr.stoptime` bits. I have some simple questions regarding the effect of those bits (this is related to #388 and #405).

> `stopcount`
> * 0: Increment counters as usual.
> * 1: Don't increment any counters while in Debug Mode or on ebreak instructions that cause entry into Debug Mode. These counters include the cycle and instret CSRs. This is preferred for most debugging scenarios.

This is quite straight forward.
But what about the _hardware performance monitors_? I assume they should be stopped when `stopcount` is set?

> `stoptime`
> * 0: Increment timers as usual.
> * 1: Don't increment any hart-local timers while in Debug Mode.

What does "timers" actually refer to (in terms of the RISC-V spec - so not talking about custom timer extensions)? Maybe `mtime` in a one-hart setup? 🤔

pdonahue-ventana, 2021-06-04:
My interpretation is that stopcount controls all the things that are controlled by mcountinhibit.  HPM is included.

I don't understand the point of stoptime because mtime is global and outside the hart.  That cannot be controllable by a hart's dcsr since different harts may have different dcsr settings, different harts may be halted at different times (and non-halted harts must still see time advance), etc.  The time CSR is inside the hart but is just a copy of the global (unstoppable) mtime.  My assumption is that stoptime=1 freezes the time CSR while in debug mode but then once you exit debug mode it will catch up to mtime.  I don't know how useful it is for hardware and/or software to temporarily freeze time as seen by abstract CSR reads or program buffer `rdtime` instructions in debug mode.  Maybe the debugger can use that to determine the time at which the hart halted?

These are just my thoughts and are not authoritative.

timsifive, 2021-06-08:
It makes sense to include hardware performance monitors in the counters that are stopped by stopcount. Is that a clarification or a feature change? I'm inclined to make a PR that says:
1: Don't increment any counters while in Debug Mode or on ebreak instructions that cause entry into Debug Mode. These counters include the cycle and instret CSRs, as well as any hardware performance-monitoring counters. This is preferred for most debugging scenarios.

Thoughts?

Once upon a time mtime was only in a CSR, and it wasn't explicitly defined to be global. stoptime comes from this time. Then later "hart-local" was added when it became clear that mtime would be global. If mtime is local (e.g. in a single-hart system) stopping it can be a nice feature to reduce the intrusiveness of debugging.

stnolting, 2021-06-09:
> I'm inclined to make a PR that says:
> 1: Don't increment any counters while in Debug Mode or on ebreak instructions that cause entry into Debug Mode. These counters include the cycle and instret [...]

Good idea! :+1: I think that was the original idea of `stopcount`.

> Once upon a time mtime was only in a CSR, and it wasn't explicitly defined to be global. stoptime comes from this time. Then later "hart-local" was added when it became clear that mtime would be [...]

I also think this is what the spec tries to say by "1: Don't increment any hart-local timers while in Debug Mode.". But it should be more specific:
* If the setup is a single-hart setup then MTIME _is_ hart-local. So it can be halted by setting `stoptime` (implicitly halting `time` CSRs).
* If the setup is a multi-hart setup that shares a single MTIME then the idea by @pdonahue-ventana sounds good: freeze `time` CSRs as long as `stoptime` is set but do _not impact_ actual MTIME counters.

pdonahue-ventana, 2021-06-09:
> It makes sense to include hardware performance monitors in the counters that are stopped by stopcount. Is that a clarification or a feature change? I'm inclined to make a PR that says:
1: Don't increment any counters while in Debug Mode or on ebreak instructions that cause entry into Debug Mode. These counters include the cycle and instret CSRs, as well as any hardware performance-monitoring counters. This is preferred for most debugging scenarios.

The new text is good but I have a minor concern about how that interacts with the old text.  It's very easy for an implementation to disable cycle/instret/hpmcounter in debug mode.  It's also very easy for an implementation to disable instret on an ebreak that enters debug mode.  In fact, ebreak is not allowed to increment instret at all (regardless of stopcount) since the privileged spec says:
> As ECALL and EBREAK cause synchronous exceptions, they are not considered to retire, and should not increment the minstret CSR

What may not be easy for implementations to do is disable all HPM counting on these ebreaks.  Taken literally, an instruction cache miss counter shouldn't increment if the ebreak instruction got an Icache miss.  That seems difficult unless all HPM information follows each instruction through the pipeline and is only written when instructions retire.  That doesn't seem like a necessary requirement and it's certainly not going to be popular.

The proposed text also loses "On single-hart cores cycle should be stopped, but on multi-hart cores it must keep incrementing."  This matters because mcycle is per-core, not per-hart.  You can't stop mcycle if only one hart is in debug mode.  You could stop mcycle if all harts are in debug mode and they all have stopcount=1.  I know that some implementations don't allow one hart on a core to be halted while another is not (and this is accomplished via hard-coded halt/resume groups).

I suggest saying:\
1: Don't increment any counters while in Debug Mode. These counters include the instret CSR as well as any hardware performance-monitoring counters. If all harts on a processor core are in Debug Mode and have stopcount=1 then the cycle CSR should stop incrementing.  Otherwise, the cycle CSR must keep incrementing.  This is preferred for most debugging scenarios.

Because of the priv spec, I think that this doesn't change the instret requirement but it alleviates the HPM issue.  It may also allow the cycle counter to increment a few extra times but the important thing is that it's disabled during the very long time spent in debug mode.

Alternatively, the existing single-hart/multi-hart description of cycle could be used (require cycle to keep incrementing even if everyone who uses it is halted and asking for it to stop counting).  I carefully used "should" so that anyone following the existing text is compliant.

As for stoptime, the history makes sense.  There should probably be some explanation about the expected behavior for systems where mtime affects multiple harts (freeze the time CSR but then it catches up later?) and where it affects only a single hart (freeze time and mtime and it doesn't catch up?).

jscheid-ventana, 2021-06-10:
Should this clarify if `stoptime` does or does not affect an `mtime` vs `mtimecmp` comparison and resultant `MTIP` value?

pdonahue-ventana, 2021-06-11:
Interrupts are disabled in debug mode.  However, MTIP is visible to the debugger in mip.

If the global mtime keeps incrementing but the local time is frozen, I can imagine different MTIP behaviors.  The CLINT-like behavior would continue to compare mtime against mtimecmp and send MTI to the hart.  An alternate implementation could have mtimecmp be inside the hart (accessible via a memory-mapped access) and could compare against time instead of mtime (which is allowable since "The time CSR is a read-only shadow of the memory-mapped mtime register.")

This stoptime mechanism violates that "read-only shadow" requirement, though only in debug mode so maybe that's OK.  But is it really useful to temporarily stop one hart's time (as visible only to external debuggers) on multi-hart implementations where other harts aren't halted?  Is it OK to have this MTIP behavior which may be inconsistent with the time CSR behavior?

I think that stoptime=1 should probably just be discouraged on implementations where the global mtime cannot be stopped (either because there may be other harts that are not halted or because of implementation choices where mtime is unstoppable).

pdonahue-ventana, 2022-05-27:
#699 clarified the stoptime description.

closed 2022-05-31 by stnolting