riscv/riscv-debug-spec #644: Which of the *context CSRs shall be implemented?

Silabs-ArjanB, 2021-06-10:
Hello,

Section 5.5.7 Hypervisor Context (hcontext, at 0x6a8) has the following two statements that seem conflicting:

> If the H extension is not implemented then this register is not implemented,

> If the H extension is not supported, it's recommended to implement no more than 6 bits on RV32, and 13 on RV64.

Is there an implied difference between 'H extension not being implemented' and 'H extension not being supported'?

Specifically, if I have a core only supporting Machine mode (without Supervisor mode support and without the H extension implemented), would it be compliant to the Debug specification to implement the following?

- hcontext (at 0x6a8) not implemented and any access attempt to this CSR will result in an illegal instruction exception.
- scontext (at 0x5a8) not implemented and any access attempt to this CSR will result in an illegal instruction exception.
- mcontext (at 0x7a8) implemented as a WARL CSR which is hard-wired to 0 (so, no illegal instruction exception upon access attempt, even though the Debug specification states this is an 'alias' to hcontext)
- mscontext (at 0x7aa) implemented as a WARL CSR which is hard-wired to 0 (so, no illegal instruction exception upon access attempt even though the Debug specification states this is an 'alias' to scontext)

pdonahue-ventana, 2021-06-11:
> Is there an implied difference between 'H extension not being implemented' and 'H extension not being supported'?

I don't believe so.  Even if the H extension is supported/implemented but is disabled via misa.H=0, the privileged spec says:
> When a standard extension is disabled by clearing its bit in misa, the instructions and CSRs defined
or modified by the extension revert to their defined or reserved behaviors as if the extension is not
implemented.

So whether it's not ever supported by this hardware or just not currently supported by this hardware, it behaves as if it's not implemented.

It's initially odd that there would be a recommendation about the number of bits when H is not implemented in a register that is not implemented.  But note that this register is also accessible as mcontext.  So the second recommendation you quoted is really talking about mcontext, not hcontext.

I tried to clarify those things in https://github.com/riscv/riscv-debug-spec/pull/646

My understanding is that your implementation is compliant.

Silabs-ArjanB, 2021-06-11:
Thank you @pdonahue-ventana , that is clear now, thanks!

closed 2021-06-11 by Silabs-ArjanB