riscv/riscv-debug-spec #652: Nomenclature suggestion: use xepc, xcause etc

JamesKenneyImperas, 2021-06-18:
To describe trap behavior when breakpoint exceptions have been delegated from Machine mode, the specification is using the imprecise term "exception PC" where it used to say `mepc`. The Fast Interrupt specification uses terminology `xepc`, `xcause` etc in this context, which I think is more precise and preserves the link with the CSR name better. I think it would improve things to adopt this naming convention.

pdonahue-ventana, 2021-06-18:
We had a short debate about this: https://github.com/riscv/riscv-debug-spec/pull/637#pullrequestreview-674761467

I'm OK with xepc and xcause.  The priv spec itself refers to xepc in 3.3.2.

timsifive, 2021-06-18:
The latest github version of the privspec does not use xepc, except in the Preface to Version 1.11.

It seems like a reasonable term to use. If we switch to it we should define it in the Terminology section.

pdonahue-ventana, 2021-06-18:
The current version uses xepc here: https://github.com/riscv/riscv-isa-manual/blob/master/src/machine.tex#L2432

timsifive, 2021-06-21:
Thanks for that pointer. My search for "xepc" didn't turn it up. In that case let's use it to be consistent with the other specs.

PR #654 (2021-07-08): xepc instead of exception PC
Fixes #652 

I don't think that we need to change mcause to xcause.  The only place I could find mcause was in the [definition of etrigger](https://github.com/riscv/riscv-debug-spec/blob/master/xml/hwbp_registers.xml#L928):
>  This trigger may fire on up to XLEN of the Exception Codes defined in \Rmcause (described in the Privileged Spec, with Interrupt=0).

It doesn't seem necessary to point to other cause registers since the codes in mcause are a superset of all xcause codes.
commit 9ac6506b (2021-07-08): xepc instead of exception PC (#654)

* Fix #587

* Use xepc instead of exception PC.
Fixes #652

* Changes based on review feedback
closed 2021-07-08 by timsifive