riscv/riscv-debug-spec #675: Tag for 1.0 stable

kito-cheng, 2021-09-23:
I saw `riscv-debug-stable.pdf` is already said it's `Version 1.0.0-STABLE`, does it possible to having a tag for that?

timsifive, 2021-09-23:
It is stable, but we are still adding clarifications and bugfixes, so the
tag would become stale. What problem exactly are you trying to solve?

Tim

kito-cheng, 2021-09-29:
The reason behind this issue is it's not easy to find version info on the github page until I download the `riscv-debug-stable.pdf`.

That's also confuse me between the `Latest stable` and `Latest release`, my first impression is, oh I should reference to last release, and seems like that's the latest version I should refer since there is tag for that.

Of cause the issue is because I am not follow the debug spec too close so I get confused about that, but maybe add few more version info in the readme.md can improve that?