riscv/riscv-debug-spec #682: tdata3 availability clarification

lbmeng, 2021-10-19:
Both current [riscv-debug-stable.pdf](https://github.com/riscv/riscv-debug-spec/raw/master/riscv-debug-stable.pdf) and the old [v0.13.2](https://raw.githubusercontent.com/riscv/riscv-debug-spec/release/riscv-debug-release.pdf) release say the following about tdata3:

`This register is accessible as tdata3 when type is 2, 3, 4, 5, or 6`

So this means type 2 trigger can support tdata3.

But experiment on SiFive Unleashed shows that when reading tdata3 it just hangs the system.

The [U54-MC core complex manual](https://static.dev.sifive.com/U54-MC-RVCoreIP.pdf) lists tdata3 as a supported CSR in chapter 12.1, but in chapter 12.2 it describes tdata3 as "Reserved" whose meaning is unclear.

Since U54-MC only implements two type 2 triggers, I suspect it does not utilize tdata3 for its functionality hence trying to read something from the type2 trigger through tdata3 will cause hardware to malfunction??

@timsifive would you confirm this hardware behavior? I believe we should clarify such in the debug spec as well.

ernie-sifive, 2021-10-19:
I can confirm that HiFive Unleashed implements v0.13 of the debug spec and does not support any triggers that include tdata3.  In 0.13, tdata3 is a WARL (optional) feature.  As required by the Architecture, when a CSR is not implemented, accessing it results in an illegal instruction trap.  That is not a "malfunction" or "hang".  Your code must be written to accommodate illegal instruction traps during the trigger discovery process.

ernie-sifive, 2021-10-19:
Again, all trigger registers are WARL.  In HighFive Unleashed (circa 2017), accesses to both tinfo and tdata3 should generate illegal instruction exceptions.  The U54 manual mentions tdata3 in a general section, but also mentions that the implementation conforms to v0.13 of the debug spec, wherein tdata3 is defined as optional.  Anyway, an access of tdata3 in HighFive Unleashed will result in an illegal instruction exception and it is up to the discovery code to handle that.

lbmeng, 2021-10-20:
Is there a way to know which trigger implements tdata3 except the trying to read and check the exception type after?

Say if tdata3 is optional for an implemented type 2 trigger, the software needs to not rely on such optional trigger functionality to provide the debug support.

timsifive, 2023-03-29:
So we need some language that makes it clear that trigger CSRs are really windows into underlying trigger registers. If a window is needed, then it's WARL. If a window isn't needed then accessing the CSR may trap.

pmundkur, 2026-10-30:
One additional item I'd like to clarify, based on the comment above that

> Trigger type 2 is allowed to have tdata3. But none of the triggers that are allowed to have tdata3 are required to do so.

Suppose multiple trigger types (e.g. `mcontrol6` and `itrigger`) are implemented for `tselect1=0`.   Can the `mcontrol6` at `tselect1=0` use `tdata3` but `itrigger` not use `tdata3`?  Both triggers have this line in their specs:

> If textra32 or textra64 are implemented for this trigger, it only matches when the conditions set there
are satisfied.

If so, if a non-zero value is written to `tdata3` (for `tselect=0`), and then `tdata1` is written to configure it as `itrigger`, does the trigger stay `disabled` for WARL reasons?  i.e. does configuring the trigger at `tselect=0` to be `itrigger` _require_ `tdata3` to be zero (with no such restriction for configuring it as `mcontrol6`)?