riscv/riscv-debug-spec #686: Triggers in a multi-core system

lbmeng, 2021-10-20:
It looks current spec does not describe how triggers should be implemented in a multi-core system.

For example, are the following configurations allowed?

Example 1:

Hart 0 implements 2 triggers, hart 1 implements 0 trigger.

Example 2:

Hart 0 implements 2 triggers, with one type 2 trigger and one type 3 trigger. Hart 1 implements 2 type 2 triggers.

Example 3:

Hart 0 implements 2 triggers, with one type 2 trigger at index 0 and one type 3 trigger at index 1.
Hart 0 implements 2 triggers, with one type 2 trigger at index 1 and one type 3 trigger at index 0.

Example 4:

Both Hart 0 and hart 1 implement exact the same configuration triggers. (I guess this is the one we expect, as this is truly symmetric)

Please clarify.

lbmeng, 2021-10-20:
Thanks for the confirmation.

Such arbitary combination might be viewed as flexiablity, but I have to say this is not friendly to software that implements debugger support in an SMP system.

pdonahue-ventana, 2021-10-21:
This isn't really different than anything else.  These are also legal:
- hart 0 implements only M mode but hart 1 implements M/S/U
- hart 0 doesn't implement the F extension but hart 1 does
- hart 0 is RV32-only and hart 1 is configurable RV32/64
- etc.

Disallowing heterogeneous configurations would be something that would have to be driven from the main architecture, not from debug alone.

pdonahue-ventana, 2022-05-27:
The debug architecture allows flexibility.  Platform standards are expected to constrain implementation choices to make things easier for software (such as debuggers).  Implementations that want to have broad software support will follow those platform standards.  Other implementations can do whatever they want with the understanding that they'll have limited appeal.  I would expect that almost nobody would make choices that make it difficult for them to get software support.

closed 2022-05-27 by pdonahue-ventana