riscv/riscv-debug-spec #690: Default accessibility of mscontext

valtrix, 2021-11-19:
Can you please clarify the accessibility restrictions for _mscontext_ CSR? Given it is encoded like a machine CSR, will an access from supervisor mode result in an illegal trap?

timsifive, 2021-11-23:
Yes, that is right. This is why scontext was created. There really is no reason to implement mscontext unless you know you have some software that relies on it.

JamesKenneyImperas, 2021-11-24:
@timsifive, the original `scontext` in this location was accessible in S mode, and the manual states that `mscontext` is an alias for `scontext` in its new location. These two things imply that `mscontext` should be accessible in S mode as well (otherwise it is not backwards-compatible, and not a strict alias). If `mscontext` is _not_ accessible in S mode, can the description of it be updated to clarify this?

Thanks.

PR #694 (2021-12-16): Specify accessibility of mscontext
Fixes #690.

This doesn't conform with what @timsifive said but I think it's correct.  I also tried to clearly discourage use of this and explain why.

@valtrix
@JamesKenneyImperas
closed 2021-12-16 by timsifive