riscv/riscv-debug-spec #691: Clarify priority of dcsr.step with respect to other debug exceptions

JamesKenneyImperas, 2021-12-08:
I think that the description of debug exception priorities should be tightened up to clarify how debug step exceptions (`dcsr.step=1`) fit when other debug exceptions are also potentially pending. I'm also wondering if the currently specified priority for `ebreak` is anomalous.

Section 4.8.1 lists these priorities, in ascending order:
priority 0: debug step
priority 1: `haltreq`
priority 2: `resethaltreq`
priority 3: `ebreak`
priority 4: trigger module action
priority 5: halt group

This implies that debug step is always lowest priority, which would mean that if `haltreq` is asserted at the point at which a debug step exception would otherwise be taken, then `haltreq` should be reported (cause 3).

However, section 4.4.1 (Step Bit in Dcsr) says this:
_If the instruction that is executed causes the PC to change to an address where an instruction fetch causes an exception, that exception does not occur until the next time the hart is resumed. **Similarly, a trigger at the new address does not fire until the hart actually attempts to execute that instruction**._

This implies that a trigger module debug exception (priority 4) on the _next_ instruction does not take priority over a debug step exception (priority 0) on completion of the _current_ instruction, which I think makes sense. This would in turn imply that if `haltreq` is asserted at the point at which a debug step exception would otherwise be taken, then `haltreq` should _not_ be reported, because if a trigger module debug exception was also asserted that would have higher priority, which contradicts the statement above.

I think it might help to redefine exception priorities to include _current_ instruction and _next_ instruction in the priority ordering, by adding the explicit rule that all debug exceptions occurring _after the current instruction_ have priority over anything that will occur _before the next instruction_. This could give this priority order:

**after current instruction**
1: trigger module action with timing After (priority 4)
2: debug step (priority 0)
**before next instruction**
3: halt group (priority 5)
4: trigger module action with timing Before (priority 4)
5: `resethaltreq` (priority 2)
6: `haltreq` (priority 1)

I have omitted `ebreak` from this list, because if that instruction causes entry into Debug mode it is going to have the effect of disabling any other attempt to enter Debug mode. This means that I think it should really sit above step 1 in the above list, and its indicated priority of 3 is really meaningless.

I think the only significant case where this really makes a difference is to clarify what happens when `haltreq` is asserted _during execution_ of an instruction when `dcsr.step=1`, which currently isn't clear.

Thanks.

pdonahue-ventana, 2021-12-09:
There's overlap between this and https://lists.riscv.org/g/tech-debug/topic/86584183

It has been on my to-do list to update the spec based on that prior thread.  What are your thoughts about that thread?

JamesKenneyImperas, 2021-12-09:
I think I'm going to close this issue. After speaking to some hardware debugger developers, I think I'm hoping for too precise a definition here.

JamesKenneyImperas, 2021-12-09:
Actually, having read @pdonahue-ventana 's linked message, I can see I'm not the only one thinking there is something here that could perhaps be defined more precisely, so I'll reopen this as a placeholder. I need to pick through that thread in detail to understand how it overlaps with this - thanks for pointing me at it!

After speaking with some debug hardware implementers, I think I'm probably hoping for a more precise definition than is ever going to be possible. We discussed the scenario where two instructions _A_ and _B_ are to be executed in program order, there is a `haltreq` pending after execution of _A_ and a (higher priority) address match trigger with Before timing on instruction _B_ which will be executed next. After executing _A_, should a debug exception be taken with cause 2 or cause 3? Their answer (I think) is that it could be either, depending on the state of the pipeline. Although, if we happened also to be single-stepping instruction _A_ (using `dcsr.step=1`) then the only legal answer is 3, because reporting a trigger module breakpoint in this case is expressly forbidden by section 4.4.1. Ugh.

Anyway, I'll leave this open for a placeholder for now in the hope that more clarity can be provided on legal behaviors.

JamesKenneyImperas, 2021-12-09:
@pdonahue-ventana, I think the priority issue for `ebreak` you raise on that thread could be addressed by considering the exception types in phases as I vaguely proposed above. In the case where `haltreq` was pending before execution of an `ebreak` instruction, then the `haltreq` would be applied _before_ the instruction started and therefore we would enter Debug mode with cause 3 without executing the `ebreak` at all. If the `haltreq` was applied after the `ebreak` had started, then the `ebreak` would win. I also think that it would make sense for an `ebreak` to win over a trigger on the `ebreak` instruction with After timing, even though the trigger is ostensibly higher priority, because by the time the trigger is processed the processor should already be in Debug state with cause 1 as a result of the `ebreak`.

I think if you look at it the right way, the priority order you are proposing for all exception types actually looks close to the one I came up with above (move my _after current instruction_ set below the _before next instruction_ set and they look very similar). I think trigger types with Before and After timing sit separately in the list, though.

pdonahue-ventana, 2023-03-28:
This is addressed in #693.

closed 2023-03-28 by pdonahue-ventana