riscv/riscv-debug-spec #697: Zicboz

pdonahue-ventana, 2021-12-22:
The [RISC-V Base Cache Management Operation ISA Extensions](https://github.com/riscv/riscv-CMOs/blob/master/specifications/cmobase-v1.0-rc2.pdf) document which was [recently ratified](https://wiki.riscv.org/display/TECH/Recently+Ratified+Extensions) says:
> This specification also recommends that the behavior of trigger modules with respect to the Zicboz
extension should be defined in version 1.0 of the debug architecture specification.

It includes suggestions for the behavior.  We should probably make the appropriate updates either as mandatory behavior (which is arguably still backward-compatible since it only affects Zicboz implementations) or as suggestions (which may be desirable if existing implementations support draft debug 1.0 and draft Zicboz).  We can also at least acknowledge/mention Zicbom and Zicbop (even if we still leave the behavior up to implementations as the CMO spec suggests for now).

timsifive, 2023-03-30:
This will be addressed by #821.

pdonahue-ventana, 2023-04-18:
#821 is now merged.

closed 2023-04-18 by pdonahue-ventana