riscv/riscv-debug-spec #710: Issues with mcontrol and mcontrol6

irichter, 2022-03-20:
I've noticed some issues with the `mcontrol` documentation, also present in the `mcontrol6` documentation. The first has a simple fix, but the latter two should be discussed.

1. In the case of an implementation supporting a lower bitmask comparison (`mcontrol.match` == 4 or `mcontrol.match` == 12), `tdata2` must be able to hold the full `XLEN` bits, otherwise it's not necessarially possible store all desired values to be ANDed with the compare value in (`XLEN`-1):(`XLEN`/2) of `tdata2`.
2. The specification does not indicate that including both Execute Address and Load Address triggers within a single chain is permitted or prohibited. To the best of my knowledge, current hardware implementations based on RocketCore and BOOM will fail to handle this chaining. (RocketCore does not currently give any ISA-visible indication of its failure here, except that the breakpoint exception never happens.) It is trivial to add the capability, and I've done so in a RocketCore prototype. The specification should require implementations not supporting this type of chaining indicate lack of support, either by zeroing the chain bit, or by zeroing the relevant non-conforming execute/store/load bits.
3. If chain that includes both Execute Address and Load Address triggers, the specification is ambiguous about what value should be present in `tval` when the exception occurs. In most implementations, the Load Address would be known later, and therefore be "the address which caused the" entire chain of "triggers to fire", but it's not clear that this is desired, and it's certainly not required by the specification. Perhaps the specification should indicate that if both an Execute Address and a Load Address triggered the chain, that the Load Address will be present in `tval`, especially since the Execute Address will end up in `epc` anyway. Alternatively, either the first or last trigger in the chain could be specified as contributing its address to `tval`, or maybe the specification should indicate this is entirely up to the implementation.

pdonahue-ventana, 2022-03-21:
> In the case of an implementation supporting a lower bitmask comparison (mcontrol.match == 4 or mcontrol.match == 12), tdata2 must be able to hold the full XLEN bits, otherwise it's not necessarially possible store all desired values to be ANDed with the compare value in (XLEN-1):(XLEN/2) of tdata2.

I wasn't involved in that definition but that's almost certainly intentional since there's obviously no way to have XLEN bits of mask plus XLEN bits of data in a single CSR.  A separate mask register (tdata4?) could have been added but I imagine that they decided to not add an additional CSR for this case and to just have people chain a match=4 trigger and a match=5 trigger to mask+compare both the upper bits and the lower bits.

> The specification does not indicate that including both Execute Address and Load Address triggers within a single chain is permitted or prohibited. To the best of my knowledge, current hardware [...]

I can't comment on those implementations.  However, the chain bit is WARL and if it's going to treat chain as if it is 0 then it seems like it should be setting it to 0 (analogous to the explicit requirement about chains that are too long).  That's just how WARL works: If you write a value that hardware can't handle then it changes the value to something that it can handle.  It's software's responsibility to read it back and see that it was changed.

> If chain that includes both Execute Address and Load Address triggers, the specification is ambiguous about what value should be present in tval when the exception occurs. In most implementations, [...]

There is no single address which caused the entire chain to fire.  In your example of chained PC and load triggers, that's obviously true.  It's also true with:
- load=1, select=0, match=0, tdata2=4, chain=1
- load=1, select=0, match=0, tdata2=5, chain=1
- load=1, select=0, match=0, tdata2=6, chain=1
- load=1, select=0, match=0, tdata2=7
- 32b load from address 4 (which accesses bytes 4,5,6,7 and, if all 4 are compare values as recommended, all 4 triggers in the chain will fire on different addresses

> Perhaps the specification should indicate that if both an Execute Address and a Load Address triggered the chain, that the Load Address will be present in tval, especially since the Execute Address [...]

The spec can't possibly comprehend all combinations of chained triggers so I don't think it should dictate special cases like execute address plus load address.  It could dictate something like first or last but that would place a burden on some implementations.  For instance, if it dictates that the last trigger in the chain is used for tval and you have:
- trigger 0: load=1, select=0, match=0, tdata2=4, chain=1
- trigger 1: execute=1, select=0, match=0, tdata2=0x102
- pc=0x100 is a 32b instruction that loads from address 4

That would require tval=0x102 which is neither the last address to be known in the pipeline nor the PC (though it is closely related to the PC).  This may require tracking additional state through the pipeline that may be burdensome for some implementations.  If we instead mandated that tval match the first trigger then the same argument applies for the case where the example's triggers were reversed.

That's why we ended up with the compromise that explicitly allows tval to be the address that caused any of the triggers to fire.  I imagine that a lot of implementations will take chained execute+load triggers and set tval to the load address as you suggest, but that's not mandated.

irichter, 2022-03-22:
> I wasn't involved in that definition but that's almost certainly intentional since there's obviously no way to have XLEN bits of mask plus XLEN bits of data in a single CSR. A separate mask register [...]

The specification [currently](https://github.com/riscv/riscv-debug-spec/blob/e9994d7174efa3af25212c2cb8aa6f15fec78503/xml/hwbp_registers.xml#L230) says that if the only valid comparisons are address comparisons (`mcontrol.select` == 0), then `tdata2` only needs to be able to hold enough bits to store an address, which is often less than `XLEN`. This should be fixed.

> I can't comment on those implementations. However, the chain bit is WARL and if it's going to treat chain as if it is 0 then it seems like it should be setting it to 0 (analogous to the explicit [...]

I agree the the implementation should change the value. I guess I'm just wondering if the specified behavior should be to zero the `chain` bit, or to zero the unsupported values among the `execute`, `load`, and `store` bits. (Might be worth not just zeroing `chain` so that the software can know what specifically is not supported.)

pdonahue-ventana, 2022-03-24:
> The specification [currently](https://github.com/riscv/riscv-debug-spec/blob/e9994d7174efa3af25212c2cb8aa6f15fec78503/xml/hwbp_registers.xml#L230) says that if the only valid comparisons are address [...]

Yes.  I think that the statement about tdata2 holding only valid addresses works for match=0, 1, 2, 3, 8, 9.  It does not work for 4, 5, 12, 13.  Or at least it does not likely work in a useful way.  See #716.

> I agree the the implementation should change the value. I guess I'm just wondering if the specified behavior should be to zero the chain bit, or to zero the unsupported values among the execute, [...]

I see what you're saying.  I don't really have an opinion on whether this should be specified when so much else related to WARL both here and in the privileged architecture is not specified.  Maybe others will express opinions.

closed 2022-03-29 by timsifive

timsifive, 2022-03-30:
> I agree the the implementation should change the value. I guess I'm just wondering if the specified behavior should be to zero the chain bit, or to zero the unsupported values among the execute, [...]

We're in a bit of a corner now where discovering what triggers are support is basically impossible. The best a debugger can do is to try to set the triggers it wants to, and see if that works. [Configuration structure](https://github.com/riscv/configuration-structure/blob/master/schema/debug-extension.asn) is the plan to deal with that.

PR #716 (2022-03-29): If masking is supported then tdata2 must be able to hold all XLEN bits.
Fixes #710
commit eeef4d58 (2022-03-29): If masking is supported then tdata2 must be able to hold all XLEN bits. (#716)

Fixes #710