riscv/riscv-debug-spec #713: Reference to configuration string unclear

christian-herber-nxp, 2022-03-22:
Among others, Section 3.14.9 Configuration String Pointer 0 refers to
`The configuration string itself is described in the Privileged Spec`
However, I cannot locate this, so I assumed this moved in the meanwhile in the spec. Maybe this needs to be adjusted

pdonahue-ventana, 2022-03-22:
The RISC-V Configuration Structure Task Group was formed 2 years ago to come up with the thing that this register will point to.
- https://lists.riscv.org/g/tech-config
- https://github.com/riscv/configuration-structure/

christian-herber-nxp, 2022-03-23:
thanks for the info. But indeed, I think the reference in the debug spec should be updated. With respect to the privileged spec, mconfigptr should be referenced, which is technically speaking also not a config string but a pointer to a config data structure.
Adding the reference to the config structure spec also makes sense

pdonahue-ventana, 2022-05-27:
This was addressed in #714.

closed 2022-05-27 by pdonahue-ventana