riscv/riscv-debug-spec #718: Clarify semantics for writing supported triggers

JamesKenneyImperas, 2022-04-01:
I've not been monitoring this specification for a while, so apologies in advance if this question has been already answered in another thread - if so, please point me at that.

On January 10th this year, text about writing supported triggers (section 5.5) was updated like this:

> As a result, a debugger can write any supported trigger as follows:
> 1. Write 0 to tdata1. _(This will result in tdata1 containing a non-zero value, since the register is WARL.)_
> 2. Write desired values to tdata2 and tdata3.
> 3. Write desired value to tdata1.

I'm not sure I understand how this should work in practice. Does this mean that each trigger now has a hidden "valid" bit that is not observable by software, and that when the trigger is in an "invalid" state the values in `tdata2` and `tdata3` can be inconsistent with the apparent value of `tdata1`? This feels a bit odd if so: it means that an apparently-programmed trigger might not actually be valid if for some reason step 3 has not been performed and it happens to be the case that the non-zero WARL value seen in `tdata1` is the same one that should have been written in step 3. This means that any software using the trigger module for its own purposes can't actually know whether a trigger is enabled or not, because the observed CSR state of an enabled and disabled trigger can be identical.

I think it would be better to either see the value 0 in `tdata1` when the trigger is in the intermediate programmable state or to introduce some other explicit value to indicate a programmable but disabled trigger (assuming value 0 is needed to indicate an absent trigger).

Thanks.

timsifive, 2022-04-01:
There shouldn't be a need for an internal valid bit.

The intent is that writing 0 to tdata1 results in a value in tdata1 that
allows any value that might be supported by any of the trigger types to be
written to tdata2 and tdata3.

If one of the supported trigger types is mcontrol/mcontrol6, then likely
tdata2 can hold any value, so writing 0 to tdata1 can simply result in type
2/6 being set in the trigger.
But you're right that if a trigger only supports itrigger and etrigger,
then this mechanism might not work, depending on which
interrupts/exceptions are actually supported.

Does this need to be updated to say that you have write 0 to tdata[1,2,3]
and then write tdata[1,2,3] in order?

Tim

JamesKenneyImperas, 2022-04-05:
Hi Tim,

I think the requirement is that if the update sequence you've described is interrupted at any intermediate point then state can always be saved and restored correctly by context switch code, so that on return from the interrupt the second part of the sequence operates as expected. I'm always a bit worried when I see step-by-step procedures that this might not be the case. If there was a distinct "programmable" state visible in `tdata1` then the sequence would clearly be interrupt safe, but I'm not sure it is otherwise (though this might be me being unable to think straight).

Thanks,

James.

timsifive, 2022-04-06:
I hadn't considered context saving at all. That is an interesting corner
case.

Let me try to list all the requirements:
1. Writing one tdata register should not require changing another one.
2. If only some of the tdata registers are written, that must not result in
a valid trigger that fires in more cases than the eventual desired result.
3. A trigger can be written using a fixed sequence of writes to tdataX.
4. Context save/restore should not need to know about trigger types.

I'm not sure how to reconcile all that. My first thought was to say that a
debugger writes tdata1 with the correct type, but disabled (e.g. by
clearing the mode bits), then tdata2, then tdata3, and finally tdata1
properly enabled. But that doesn't work with #4. It could work to have a
trigger_enable bit somewhere, that is automatically cleared whenever tdataX
is written. But then where do we put that? And that's definitely backwards
incompatible.

I'm just brainstorming a little. We'll talk about this at Friday's meeting,
but I'd love to hear emailed suggestions before then.

Tim

pdonahue-ventana, 2022-04-07:
> Does the concept of SBI apply on M-mode only systems?

No.

If this M-mode-only system has cooperative multitasking then there's no problem (as long as it doesn't yield in the middle of changing a trigger).  If it sets mstatus.MIE=0 while it's modifying the triggers then there's no problem.  But if it has preemptive multitasking and it just wants to modify things without changing interrupt enables then there's a problem.  I don't know if the MIE solution is acceptable or not.

pdonahue-ventana, 2023-03-28:
This should be addressed by #721.

closed 2023-03-28 by pdonahue-ventana