riscv/riscv-debug-spec #726: Quick access to current and former stable versions

midnighter95, 2022-05-05:
Hello, 

I found there are so many 1.0.0 version specification which are all marked stable but very in date.  Is there any truly stable version that we can work on without chasing the latest one?

In addition, I found it’s quite hard to locate the stable version of 0.13 and 0.14 from the commit log. Could you please add version tags to specific commits?

Thanks.

midnighter95, 2022-05-10:
Hi Tim,
I really appreciate your efforts to make the specification compatible. But I still wonder if it’s possible to tag a commit if the spec is ratified. It could be a great help for the hardware design and verification to have a 1.0.0 version that won’t ever change. 
Thank you.

timsifive, 2022-05-10:
The currently ratified spec (0.13) is in a separate branch (called release). This is to enable bugfixes. It's a little more flexible than a tag, but the same idea. When 1.0 is ratified, we'll do something similar.

midnighter95, 2022-05-10:
@timsifive Thanks for your clarification!

closed 2022-05-10 by midnighter95