riscv/riscv-debug-spec #737: Suggesstions in release strategy

midnighter95, 2022-06-13:
Hello,
I have several suggestions in optimizing the release strategy. 
Firstly, I think it could be better to point out in the readme which version we are working on(1.0.0 stable of course).
Secondly, I propose that we add a branch for the version 0.13 ratified where we can add CI and fix typo for it.
Finally,I’m wondering if it's possible to reorganize those branches. I suggest that maybe we can retire the “release” branch.
Thank you.

timsifive, 2022-06-13:
> Firstly, I think it could be better to point out in the readme which version we are working on(1.0.0 stable of course).

Sounds good. Can you make a PR?

> Secondly, I propose that we add a branch for the version 0.13 ratified where we can add CI and fix typo for it.

It makes sense to rename the `release` branch to 0.13, because there needs to be a place for people to find 0.13 even after we release 1.0. But we do want a fixed URL that we can point people to that always contains the latest release. Can we use github releases for that?

sequencer, 2022-06-15:
Sure.

sequencer, 2022-06-16:
> It makes sense to rename the release branch to 0.13, because there needs to be a place for people to find 0.13 even after we release 1.0. But we do want a fixed URL that we can point people to that always contains the latest release. Can we use github releases for that?

Hi, @timsifive, would you mind create two branches: `v0.13-release` and `v1.0-release`, so that I can work on the CI to push to the corresponding branches.

timsifive, 2022-06-16:
I've created `v0.13-release`. I prefer not to create `v1.0-release` until we actually release 1.0. Right now `master` effectively has this role.

pdonahue-ventana, 2022-06-16:
That seems sensible to me since the README now says "The current master branch is v1.0.0-stable" and there is no release version of 1.0 yet.

sequencer, 2022-06-17:
> My concern is if we create v1.0-release that some changes will go there, and some will go to master, and we'll have to fix that up. It seems simpler to continue using master for all development for now.

Sorry, that's my fault, I meant `v1.0.0-stable`, but might have a typo...