riscv/riscv-debug-spec #742: Command structure in driver/software

markusdd, 2022-07-08:
Hi, 
I have been reading this debug spec 1.0.0 now and while I very much agree with the features outlined in section 1.5 and with the multi-hart debug strategy, I have some questions and suggestions aber the register and command structure:

1) The register offsets from DM-base increment by one, although they are 32-buit wide. As most systems will use byte-adressing, this does not play well e.g. with an APB- or generally bus-based DM register file. You would expect increments by 4 then.

2) Is it possible to emulate the command structure e.g. in the openocd driver? We are very much deviating away from using JTAG-chains within the SoC because they are all kinds of trouble with regard to power domain switching etc.
We much prefer having a central JTAG tap which then translates onto AHB/APB and from ther you can go wherever you want. Another alternative for bus access are SPI interfaces, UARTs and the like.
In our debug reg file the core registers are all visible directly by seperate addresses in the APB window without having to first write a command etc. . This is easy, cheap (no extra flops), and writing is just done by writing onto the respective register, which then gets forwarded inside the core directly.
So while all this command stuff is nice for being generic, in our case it makes 0 sense.
If we wish to be compliant: Can these command be just emulated in the e.g. OpenOCD driver? Same goes for e.g. the resets. The SoC already has provisions for that which make no sense to be replicated in the DM or hart again.

So what I am proposing is e.g. in our case: Spec compliant Debugger sends a read command for register 15. In teh driver this just gets translated into a JTAG2AHB access, reading regsiter 15, and then returning the value together with the ready status bit.

In my view, it would make a lot of sense to allow that.

3) I'm not sure if I missed it, but is there a general provision to advertise the base address in a memory-mapped DM? I know there is register 0x11 to advertise some of the features, but if you e.g. have multiple DMs in different subsystems, how do you advertise their base addresses?

markusdd, 2022-07-08:
a) I think the spec should be completely transport agnostic. If someone wants to sent debug command via bluetooth, that should also work^^ (exaggerating to make the point). Nothing should be defined for that at all.

b) I do not get how this would work in practice. The hardware will behave just like described, but the way the chip is accessed will always be SoC specific. Even when connecting e.g. to standard ARM cores always e.g. the specific cores need to be given to make clear to the host what can be used and how the DAP works. Also the Interface needs to to be given (usually SWD or JTAG) So essentially there is always a driver involved (it doesn't have to be in OpenOCD, but essentially every MCU on this planet has a specific debugger definition, already because commands to flash Firmware to non-volatile memory etc differ materially)

c) I would strongly suggest to think again about the hardware impact of the current spec. If you are targeting a small 32-bit 32-register core for low power or even a 16-register RVE one, this debug spec introduces a significant overhead without a specific need. An ARM CM0+ has less than 1k Flip-Flops, debug access port included. With this debug spec this would be close to impossible to achieve. So I strongly believe at least the option to  transfer some of the logic into debug drivers and use more efficient provisions on-Chip would be advisable.
Another way of handling this would be to offer 2 different compliance levels to this debug spec. Level 1 would offer 'feature compliance' (essentially section 1.5), Level 2 'feature compliance and control interface compliance'. Level 1 then simply needs to ship their own drivers to work 'plug'n'play, but at least all debug plugins for editors, IDEs etc would work.
Level 1 is essentially how all RISC-V debugging works today because there is no stable spec yet.

d) I might have overlooked that, but I lack some deeper spec on how breakpoints and data breakpoints (also known as Watchpoints) should behave. For breakpoints there is usually a common understanding that the instruction where the breakpoint sits will NOT be executed. For watchpoints there is no real standard. I have seen halts after the data transfer (read or write) has happened, but also halts upon detection that the transfer would be issues now but the system halts before that.
Is there any preference?

markusdd, 2022-07-11:
Thank you for the elaborate response!

Just as a preface: Our original debug implementation which we also shipped in silicon dates back all the way to early 2018, with implementation in 2017, so even before the preliminary debug spec was ratified. So this is where I am coming from with my comments above.

a) I was just joking about Bluetooth to make point ;)

b) We do use JTAG as well, but apart from having a very basic set of registers and modes, JTAG does not really encompass any additional functionality. We for example do have a JTAG2AHB bridge in our JTAG-TAP controller which gives you access to the bus system, to which our APB Debug register file for the RISC-V is attached. But the way in which this JTAG2AHB bridge is accessed and operated is specific to our implementation. I know ARM has this in their DAPs as well. So if you wish to perform e.g. any memory mapped accesses through the debugger, you need a driver for that specific SoC. I do not think that this can be generalized. The number of different TAP implementations out there is huge, and they are not only driven by RISC-V debug needs but by many other factors on the SoC as well. Having an additional TAP implementation for each and every core seems wasteful.
 
So if we wish to use APB for our DMI, how are we supossed to:
- advertise the baseaddress to the debugger? (I understand this is lacking in the concept from one of your inputs above)
- define the access to the bus system (AHB/APB does not go off-chip, so you need some way in. In the past we have used SPI for that, nowadays we use the aforementioned JTAG2AHB-bridge in our main system TAP)
- define a possible chip specific auth procedure to unlock access from outside (which might be completely unrelated to RISC-V but driven by the whole SoC requirements)
- integrate with the rest of the system that might use other cores as well (I participated in SoCs with no less than 27 cores from 3 different vendors) or has special security requirements, this e.g. ties into the whole reset discussion, where an alternative location for the respective reset controls might be unavoidable (e.g. in a security hardened system you might need to aquire a hardware semaphore to access system controls like this)

To try and make my point a bit more crisp: From the given spec I have a hard time seeing how to comply with that spec without scrapping the whole bus-attached/memory-mapped approach altogether, losing all the benefits of it, and writing a completely new TAP controller just solely for the debug register file, adding significant complexity compared to what we had before, which had already like 80% of the functions from chapter 1.5 .

This might come across as a little bit dumb, but somehow I feel this spec might still lack some provisions to really accomodate the wide range of possible implementations and core integrations out there in heterogenous systems which are not primarily defined by the fact that they contain a RISC-V core.

My biggest confusion is really the access to the DM. Without the registers being located directly in the System TAP, I see no way to advertise how to get to those registers in a byte-addressed, memory-mapped system.

As a last point: I do not get your reference to 'timing discussions in mcontrol6'? What discussion are you referring to? Is there a link you can provide?

Thanks again for taking the time to discuss this. We very much wish to adapt to the specification, but as you see there are still some gaps in understanding I guess^^

bruceable, 2022-07-11:
>    1. I'm not sure if I missed it, but is there a general provision to

Markus,

I work for SiFive and we have implemented an APB option for the DM.  If you
send me your email address I can forward you our memory-mapped
documentation.

My email address is ***@***.***

--- Bruce

timsifive, 2022-07-13:
@markusdd, it sounds like you are happy to implement the Debug Module, but don't want to use JTAG to access it. That is valid and allowed by the spec. The DTM chapter says:
> An implementation can be compliant with this specification without implementing any of this
> section. In that case it must be advertised as conforming to “RISC-V Debug Specification 1.0.0-
> STABLE, with custom DTM.” If the JTAG DTM described here is implemented, it must be
> advertised as conforming to the “RISC-V Debug Specification 1.0.0-STABLE, with JTAG DTM.”

In that case you're making up your own way to access the DM, and whatever you make up is fine. Of course existing software will have to be modified to work with it. You could come up with a way to do so over Bluetooth or any transport you care to use. Ideally then you work with this group to get a DTM for your transport defined and added to the spec, so that other people who want to use the same transport will do it in the same way. This can include a way to advertise the base address to the debugger, authentication, etc.

> As a last point: I do not get your reference to 'timing discussions in mcontrol6'? What discussion are you referring to? Is there a link you can provide?

See the "Match Control Type 6" section in the spec, which talks about the timing bit.

timsifive, 2022-07-14:
So you have a DM buried somewhere in your SoC. The spec currently defines only one method for an external debugger to talk to a DM: JTAG. (It explicitly allows you to use something else, as long as you're explicit about it.) We're open to adding other methods to talk to the DM, such as using a standard JTAG2AHB bridge and APB (or some other bus?) behind that. Discovery would be part of that method.

markusdd, 2022-07-14:
yes, 'buried in our SoC' is the right word I guess^^

Let's say the RISC-V lives in a subsystem that occupies the address space under 0xF000_0000 and the AHB2APB decoder in that subsystem lives at 0xF100_0000. Then the DM would be one of the APB-peripherals under that decoder, let's say at
0xF100_3000 (commonly nowadays you reserve 4kB - so 12bit - for each APB peripheral).

So from the current spec, register 0x00 of the DM would have address 0xF100_3000, the feature register placed at word-addr 0x11 from the spec, would live at 0xF100_3000 + (0x11 << 2) = 0xF100_3044.

So in the end, the dicovery method would need to know
a) how to operate the JTAG2AHB to do reads and writes on the bus (usually a defined protocol of selecting the JTAG2AHB bridge by an initial IR scan code and then a series of DR scans to perform a read or write) 
b) how to find the 0xF100_3000 baseaddr

As soon as you go to memory mapped the actual bus protocol does not matter to the outside debugger. It still only plays JTAG sequences to deliver addresses and read/write commands, if the internal bus is wishbone, AHB, AXI or whatever does not really matter. Also, for convenience, if the Memories of the RISC-V are also bus accessible a simple advertisement of their size and baseaddrr is enough. You do not need any further provisions in the DM, the debugger can just access and read/manipulate them directly if needed. Same goes for the reset by the way: If there are already reset controls in the system, which basically applies to most SoCs, there is no inherent need to have them again in the DM. Just advertise the location and the debugger can request a reset directly there (for us there is central location in each subsystem for that but that might differ materially per SoC).

Of course, to really be generic, such a bus bridge could also be realized using SPI (we had that in the past) or UART with flow control.

ARM has circumvented some of this by forcing their reset vector to address 0 and having many fixed addresses cluttered all over the 32-bit address space. But I consider this to be an actual weakness, because now with multiple cores you have an individual view of the SoC from each core. And if you need to go to the memory of anotehr core, you need to have an alias window, cluttering the address space even more.
There is no need to do that with RISC-V. Our core can be configured to have its reset vector wherever it wants. But to retain that flexibility also the DM need to be able to move within the memory map.

I hope this all makes sense.

timsifive, 2022-07-18:
That all sounds reasonable. I'd add that the specified JTAG DTM is also simply a bus bridge.

markusdd, 2022-07-18:
Yes indeed it is. It just translates from JTAG to AHB via a JTAG data register via a specified command structure. (ARM DAP also has this functionality, but very basic, ours can do a bit more in terms of speed)

timsifive, 2023-03-29:
Closing due to lack of activity. I'd still love to add more debug transports to the spec, but not in 1.0.

closed 2023-03-29 by timsifive