riscv/riscv-debug-spec #744: Privilege mode when in debug mode but not executing an abstract command?

Silabs-ArjanB, 2022-07-25:
The Version 1.0.0-STABLE version of the debug specification (as well as the 0.13.2 version states the following in 'section 4.1 debug mode':

> 
> When executing code due to an abstract command, the hart stays in Debug Mode and the following apply:
> 
> 1. All operations are executed with machine mode privilege,

I am wondering whether that should be stated as: 

> When executing code due to an abstract command, the hart stays in Debug Mode. When in Debug Mode the following apply:
> 
> 1. All operations are executed with machine mode privilege,

If not, at what privilege mode is code being executed while in debug mode, but not executing code due to an abstract command (e.g. because other code of the park loop is being executed or because code from a program buffer is being executed)?

pdonahue-ventana, 2022-07-25:
> If not, at what privilege mode is code being executed while in debug mode, but not executing code due to an abstract command (e.g. because other code of the park loop is being executed or because code from a program buffer is being executed)?

Architecturally speaking, code is only executing while there's an abstract command in progress.  The park loop is one possible implementation and is microarchitectural (which is why it's only mentioned in an appendix).  Architecturally, normal execution stops upon entering debug mode, the hart tells the DM (either via a park loop or via some other mechanism such as dedicated wires) that it has halted, and the DM tells the hart (either via the hart polling in the park loop or via some other mechanism such as dedicated wires) when there's a command to execute or when to resume.

Although the appendix suggests that the park loop runs in debug mode, it could presumably run in some mode other than debug mode as long as it has sufficient privileges to write to the memory location within the DM to indicate that it is halted, has sufficient privileges to read from the DM flag that passes commands to the hart, and enters debug mode before executing the abstract command.

That's all a bit of theory.  All of that being said, I'll now stop talking about theory and now talk about real-world implementations.  The park loop (which I assume everyone runs in debug mode as the spec suggests) needs to have sufficient privilege to access the DM locations where it's storing the "I'm halted" flag and reading the "what do you want me to do" flag.  If abstract CSR accesses are supported then it has to be able to access M mode CSRs.  So the park loop portion of debug mode has to run with M mode privilege as you say.

Your clarification doesn't require the typical park loop implementation but it accommodates one.  So to summarize this very long answer: Your proposal seems fine to me.

Silabs-ArjanB, 2022-08-03:
Hello @pdonahue-ventana 

Thank you very much for your answer above. I hope you can help me with a related questions as well.

From the description of the stopcount bit in the dcsr CSR it is very clear that instret shall be stopped when stopcount=1. That seems to be the obvious case however. Based on above discussion and based on the following statement in the Debug specification I wonder if therefore instret shall not increment either when stopcount=0 and the hart is in debug mode:

> Debug Mode is a special processor mode used only when a hart is halted for external debugging.
> Because the hart is halted, there is _no forward progress in the normal instruction stream_.

Would you agree with the following?

- _instret_ shall not increment in debug mode _independent_ of the value of stopcount (because no progress is made in the normal instruction stream). 
- _cycle_ behavior _does_ depend on the value of stopcount (as described in the Debug Specification), specifically _cycle_ will continue to increment when clocking the hart while in debug mode and stopcount=0

Best regards,
Arjan

pdonahue-ventana, 2022-08-03:
If we're talking pure theory, I would say that instret should increment in debug mode if and only if stopcount=0 and the instruction being executed is in the program buffer.  In a world of existing implementations, I think that instructions in the park loop would also increment instret when stopcount=0.

Either:
- the software being debugged doesn't care about the counters.  In that case, it probably doesn't matter whether stopcount is set or what the hardware does.
- the software does care.  In that case, the debugger should have set stopcount=1 since it wouldn't want to modify instret even by counting a few program buffer instructions.  Then, for instance, software can still do instret/cycle to calculate IPC without skewing the result from having long periods in debug mode.

I think that stopcount=0 is not desirable in general usage.  It might be useful for some really obscure things.  But stopcount=1 is the "normal" mode as suggested by https://github.com/riscv/riscv-platform-specs/blob/main/riscv-platform-spec.adoc#debug (which allows it either to be programmable with a reset of 1 or to be hardwired to 1 but does not allow it to reset to 0).

pdonahue-ventana, 2022-08-03:
I should add that I do unequivocally agree with your statement about cycle behavior.

Silabs-ArjanB, 2022-08-03:
Thank you very much

closed 2022-08-03 by Silabs-ArjanB