riscv/riscv-debug-spec #745: Access of "PC" through dpc

Silabs-ArjanB, 2022-08-03:
The debug specification section about the 'access register' abstract command states:

> The “PC” can be accessed here through dpc.

and

> dpc may be used as an alias for PC if this command is supported on a non-halted hart.

Is it defined exactly what "PC" means in the context of accessing it on a non-halted heart. For example, is it allowed to report the program counter from early in the CPU's pipeline, which typically is speculative, or shall it be the program counter of instructions that actually retired (so excluding for example the PC of an ecall or ebreak instruction as those are never considered to retire), or some other definition of 'program counter'.

pdonahue-ventana, 2022-08-03:
Hopefully it's understood that (regardless of whether register renaming is employed) accessing x3 should return the architectural x3 and not a speculative x3.  Similarly, the pc (accessible through dpc) should return the architectural pc (i.e. some pc value that could possibly have ended up in dpc if you halted before doing the read).

Silabs-ArjanB, 2022-08-03:
@pdonahue-ventana I now realize it would indeed never make sense to report some speculative value; apologies for even asking.

closed 2022-08-03 by Silabs-ArjanB

pdonahue-ventana, 2022-08-03:
No problem.  I've certainly asked my share of questions that I later realized didn't make sense.