riscv/riscv-debug-spec #747: Why do triggers in a chain have priority?

YenHaoChen, 2022-08-16:
Hello,

The specification states that triggers in a chain have priority. However, It also says that only the final (last) trigger of the chain can be taken, and others are ignored.
The specification seems inconsistent to me. Can you help clarify this?

[image]

[image]

Thank you
Best regards,
YenHaoChen

pdonahue-ventana, 2022-08-16:
The action field on the last trigger is used.  The action fields on other triggers in the chain are ignored.  But all the other fields on the other triggers are not ignored.

Example: If you want to trigger only on a load to address X where the load has pc Y then you can chain two triggers.  The priority will be the load trigger priority (since that's lower than the PC trigger priority) and the action taken will be based on the action field of whichever trigger is last in the chain.

YenHaoChen, 2022-08-16:
Thank you. Understood.

closed 2022-08-16 by YenHaoChen