riscv/riscv-debug-spec #749: expected behavior when action=1 and dmode=0

neelgala, 2022-08-26:
The spec says when `action=1 then Enter Debug Mode`. And it also says `This action is only legal when the trigger’s dmode is 1`. but doesn't specify what should happen if action=1 and dmode=0? Should the trigger firing be ignored or do we take a breakpoint exception - or something else ?

pdonahue-ventana, 2022-08-26:
It's WARL so it's impossible to have action=1 and dmode=0.

It's kind of like D and F in misa.  The priv spec doesn't say what happens when misa.D=1 and misa.F=0 because that situation cannot happen.  If you try to write that value then the WARL logic will fix it to be some legal combination (F=0, D=0).

neelgala, 2022-08-27:
got it thanks.

closed 2022-08-27 by neelgala