riscv/riscv-debug-spec #751: should setting the hit-bit cause trigger actions ?

neelgala, 2022-09-06:
The spec says `the hardware sets it when this trigger fires. The trigger’s user can set or clear it at any time`. 
Would performing a trigger action whenever the `hit` bit is set be a valid behavior ? or Should the hit bit be treated more like a status bit causing no other changes?

timsifive, 2022-09-06:
The `hit` bit indicates that something happens. It doesn't trigger any behavior.

IIRC the reason it is r/w has to do with context save/restore in complicated systems.

pdonahue-ventana, 2022-09-07:
> Would performing a trigger action whenever the hit bit is set be a valid behavior ?

Does "whenever the hit bit is set" mean "whenever the hit bit transitions from 0 to 1" or "whenever the hit bit is 1"?

If you mean the former, I think that the act of firing is what both causes the hit bit to transition to 1 and to perform the action.  So I don't know if there's really a distinction to be made unless I'm forgetting a corner case.

The latter will definitely not work for things like action=4 where the trigger fires, hit becomes 1, the trace unit is told to do a trace-notify, and the hit bit just remains 1 until somebody later comes along and decides to clear it.  You don't want to tell trace to trace-notify every single instruction retire from that time forward.

neelgala, 2022-09-07:
thanks.. for inputs. Doubt clarified.

closed 2022-09-07 by neelgala