riscv/riscv-debug-spec #753: Effect of clearing dmcontrol haltreq bit

Silabs-ArjanB, 2022-09-15:
The dmcontrol haltreq description states:

> Writing 0 clears the halt request bit for all currently
> selected harts. This may cancel outstanding
> halt requests for those harts.
> Writing 1 sets the halt request bit for all currently
> selected harts. Running harts will halt whenever
> their halt request bit is set.

I interpret the word 'may' as that it is implementation defined whether the 'actual halt request to the CPU' (not the 'bit') is cleared as well upon writing 0. In our CPU we make the halt request sticky such that a 1 cycle high pulse on the 'halt request bit' cannot be missed by the CPU. We then clear this sticky halt request when debug is entered due to a halt request. 

Is this interpretation of above dmcontrol haltreq text compliant with the Debug specification? 

Is there a 'preferred' interpretation of above text? E.g. opposed to our implementation we could also remove the 'stickyness', but then it cannot be guaranteed that debug mode is entered for a short high pulse on haltreq (which might not be compliant to the statement 'running harts will halt whenever their halt request bit is set'). 

(I understand that the debugger could use dmstatus to judge whether it actually halted a hart or not, but I do not know if all debuggers will rely on that 'handshake' or whether a debugger might assume that even a 1 cycle haltreq pulse is guaranteed to lead to eventual debug entry).

pdonahue-ventana, 2022-09-16:
> I interpret the word 'may' as that it is implementation defined whether the 'actual halt request to the CPU' (not the 'bit') is cleared as well upon writing 0.

That is my interpretation, as well.

> In our CPU we make the halt request sticky such that a 1 cycle high pulse on the 'halt request bit' cannot be missed by the CPU. We then clear this sticky halt request when debug is entered due to a halt request.

Do you just capture posedge haltreq or is it level-sensitive?  Do you clear the bit on the cycle when debug is entered or every cycle that you're in debug mode?  If you treat them both as level-sensitive then I'm pretty sure it will work.  I assume that's what you're doing, but just to think through the other case: If you just detect edges then I also think it will work because "resumereq is ignored if haltreq is set."  If haltreq is still set then there's no way to exit debug mode so there's no need to still see haltreq as asserted.  Therefore, you only need to capture the posedge of haltreq and you know that there was a negedge before exiting debug mode.  This would presumably allow haltreq to be delivered similar to a message signaled interrupt.  It's less straightforward if you assert hartreset while in debug mode and keeping haltreq asserted (which should reset and then halt before executing the reset vector), though I think that it's not terribly complicated to handle that (though the MSI approach may not work well).

> (I understand that the debugger could use dmstatus to judge whether it actually halted a hart or not, but I do not know if all debuggers will rely on that 'handshake' or whether a debugger might assume that even a 1 cycle haltreq pulse is guaranteed to lead to eventual debug entry).

If the debugger clears the haltreq before the hart is halted then presumably it doesn't want it to halt anymore.  The hart might still happen to halt depending on race conditions or implementations such as yours (in which case the debugger would just send resumereq immediately since it doesn't want to debug anything).  I don't really know why a debugger would change its mind like this so I don't know if it really matters.

It's like runt pulses of level-sensitive interrupts which the CPU is allowed to ignore (and which may be bad for performance to actually trap since the handler will see no pending interrupt and will simply return, having wasted a bunch of time trapping and returning for no useful purpose).

I can't speak for all debuggers but [OpenOCD](https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L4194) waits before clearing haltreq.  (If it doesn't halt within the timeout then it raises an error and does not clear haltreq.)

Silabs-ArjanB, 2022-09-17:
@pdonahue-ventana 

>> Do you just capture posedge haltreq or is it level-sensitive? 

We treat the haltreq from the debug module into the core as a level sensitive signal, but additionally we make it sticky (a flop will get set whenever haltreq = 1).

>> Do you clear the bit on the cycle when debug is entered or every cycle that you're in debug mode?

We clear the sticky version of the signal whenever debug is entered _because_ of the halt request (we do not keep on clearing it when we are in debug mode).

It sounds to me then that the above is within spec.

pdonahue-ventana, 2022-09-17:
> We clear the sticky version of the signal whenever debug is entered because of the halt request (we do not keep on clearing it when we are in debug mode).

If you treat haltreq as level-sensitive then I don't think this will work.  Here's the normal sequence:
- debugger sets haltreq
- halt happens
- debugger observes that you're halted
- debugger clears haltreq

That means that haltreq will be set for some period of time while you're in debug mode.  Your implementation clear it upon entering debug mode but you'll set it again the very next cycle because haltreq is still asserted (and will be for many more cycles).  The hart will probably halt much faster than the debugger can issue another command to clear it, so haltreq is almost guaranteed to still be 1 for some period of time after entering debug mode even if the third step is omitted.

I think that you need to set the flop whenever haltreq is 1 (level-sensitive as you have) and clear the flop whenever you're in debug mode (level-sensitive instead of your edge-triggered).  Or don't have the flop at all since, like interrupts, you don't need to guarantee debug mode entry if there's a runt haltreq pulse.

Silabs-ArjanB, 2022-09-19:
Hello @pdonahue-ventana 

>> Or don't have the flop at all since, like interrupts, you don't need to guarantee debug mode entry if there's a runt haltreq pulse.

If we do not need to guarantee debug mode entry if there's a runt haltreq pulse, then things get easier indeed.

I (mis)read the specification part that says "Running harts will halt whenever their halt request bit is set." as implying that the CPU would not be allowed to miss haltreq pulses.

pdonahue-ventana, 2022-09-19:
> I (mis)read the specification part that says "Running harts will halt whenever their halt request bit is set." as implying that the CPU would not be allowed to miss haltreq pulses.

That's interesting and maybe I'm wrong.  @timsifive: What is your interpretation of this?  If there's a runt pulse where the debugger sets haltreq and then clears it incredibly quickly, is the hart required to halt?

timsifive, 2022-09-29:
> If there's a runt pulse where the debugger sets haltreq and then clears it incredibly quickly, is the hart required to halt?

No. It's not useful behavior. There already is one halt bit per hart, so if the debugger wants to keep trying to halt, it can keep that bit set. However, clearing the haltreq bit does not guarantee the halt request is aborted.

Silabs-ArjanB, 2022-09-30:
@timsifive , @pdonahue-ventana Thank you both for your responses. It is clear now how haltreq is supposed (and allowed) to behave.