riscv/riscv-debug-spec #754: What is the purpose of restricting on setting mcontrol.dmode and mcontrol.chain?

YenHaoChen, 2022-09-15:
Hello,

The debug specification states two constraints on setting the dmode and chain.
1. If the next dmode=1, writing dmode to 0 must set the chain to 0.
2. If the previous dmode=0 and chain=1, writing dmode to 1 should be ignored.

[image]

I have two questions about the constraints.
1. What is the purpose of those two constraints? Do those constraints guarantee any property on the trigger chain (e.g., all trigger chains will have the same dmode in the chain, and M-mode can never break a trigger chain of D-mode)? 
2. How does a debugger set up a trigger chain with those two constraints? Is the debugger expected to use special tricks to avoid clashing with the constraints (e.g., set up all dmode with chain=0 in the chain first and then set up the trigger)?

Best regards,
YenHaoChen

pdonahue-ventana, 2022-09-16:
If dmode=1 then the trigger is only writable by the external debugger.  If dmode=0 then software on the hart can program the trigger.  To some extent, dmode is a security mechanism that prevents a native debugger from interfering with the external debugger.

Let's take each constraint separately:
> If the next dmode=1, writing dmode to 0 must set the chain to 0.

If the next dmode=1 then the external debugger owns it.  If there is a write to this trigger with dmode=0 (either by the external debugger changing dmode from 1 to 0 and relinquishing the trigger or by software just doing normal trigger programming) then this trigger is now owned by hart software.  If this trigger becomes writable by the native debugger then you'd have problems if you chained it with an external debugger trigger.  The outcome would be either undesirable or a security problem.

Basically, the entire chain needs to be owned by the same debugger.

> If the previous dmode=0 and chain=1, writing dmode to 1 should be ignored.

Both triggers are owned by the native debugger and they're chained.  Then there's a write of dmode=1 so the external debugger now wants to own this one.  As above, we don't want to have half of the chain owned by one debugger and the other half owned by another debugger.  The options are to clear the chain bit on the previous trigger or to ignore the dmode write to this trigger.  This says to do the latter.

> How does a debugger set up a trigger chain with those two constraints?

Native debuggers always write dmode=0 (and, in fact, are only able to write that value).  External debuggers are expected to use action=1 (since they would normally not want to program a trigger to pass control to a native debugger which is probably not running).  To do so, they must always write dmode=1.  If external debuggers use trace actions or custom actions then I would also expect them to use dmode=1 simply to prevent anyone from interfering with those triggers.  The above constraints only affect the cases where triggers are being dynamically reallocated between external and native debuggers.

YenHaoChen, 2022-09-19:
Thanks for the reply. With your information, I have a more precise understanding of the use scenario of triggers.

1. Native debuggers always write dmode=0, and external debuggers are expected to write dmode=1.
2. The entire chain needs to be owned by the same debugger; otherwise, the outcome would be either undesirable or a security problem. In other words, we don't want to have half of the chain owned by one debugger and the other half owned by another debugger.
3. The two constraints prevent native and external debuggers from interfering with each other.

However, the native and external debuggers could still claim partial ownership of a trigger chain owned by the other previously with those two constraints. For example, the native debugger could write the tail of an external debugger's chain to claim partial ownership. Additionally, the external debug could write the head of a native debugger's chain to claim partial ownership. The above cases will result in half of the chain being owned by one debugger and the other half being owned by the other.

I guess the spec wants to allow internal/external triggers to use all triggers, even if owned by another previously. This seems to conflict with the fact that we don't want half of the chain owned by one debugger and the other half owned by another.

Is there any plan to revise the constraints to guarantee that the same debugger will own the entire chain to prevent undesirable outcomes and security problems? Or, will there be any method to delete the whole trigger chain owned by the other debugger?

pdonahue-ventana, 2022-09-19:
Practically speaking, it would be strange and probably unpleasant to have two debuggers running at the same time.  Perhaps the main reason for the dmode field is to prevent a native debugger from accidentally writing action=1 which could cause the hart to halt.  With no external debugger, there's no way to resume and this buggy software would hang.

But maybe somebody wants to use an external debugger to debug the native debugger?  It's theoretically possible to have two simultaneous debuggers using nonoverlapping resources, so...

> However, the native and external debuggers could still claim partial ownership of a trigger chain owned by the other previously with those two constraints. For example, the native debugger could write the tail of an external debugger's chain to claim partial ownership.

I don't think that's possible.  The external debugger would set dmode=1 on that trigger ("external debuggers will always set this bit when configuring a trigger"), preventing the native debugger from writing to it.

It would be possible for the external debugger to do such a write but the spec says that debuggers must check the chain field on the previous trigger if they're writing mcontrol/mcontrol6.  There's no hardware mechanism to prevent a buggy external debugger from trashing its own triggers and I don't think such a mechanism is even possible.

> Additionally, the external debug could write the head of a native debugger's chain to claim partial ownership. The above cases will result in half of the chain being owned by one debugger and the other half being owned by the other.

The external debugger can always overwrite any of the native debugger's triggers (the head/middle/tail of a chain or single triggers that are not in a chain at all).  The external debugger has access to everything (triggers with dmode=0, CSRs, all of memory, etc.) and, as above, it's expected to not modify things that it shouldn't modify.

> Is there any plan to revise the constraints to guarantee that the same debugger will own the entire chain to prevent undesirable outcomes and security problems? Or, will there be any method to delete the whole trigger chain owned by the other debugger?

I don't think there's any plan to change the constraints since there are already a number of constraints on both hardware and software (native/external) to prevent undesirable outcomes.  Native debuggers cannot modify external debugger triggers because of dmode.  External debuggers can inherently modify everything and are expected to use that power responsibly (much like M mode code in the non-debug world).

pdonahue-ventana, 2022-09-20:
You stated these two rules which accurately summarize what I said:
> 1. Native debugger (writing dmode=0) cannot overwrite the external debugger's trigger chain (dmode=1).
> 2. A hardware mechanism can't prevent a buggy external debugger. Thus, the external debugger is allowed to access everything with responsibility.

And you have two concerns.

First, I think that you're concerned about this:
> If the next dmode=1, writing dmode to 0 must set the chain to 0.

Neither of the two rules affects this case.  Consider this scenario:
- trigger N has dmode=0 and chain=0 initially
- trigger N+1 has dmode=1 initially
- native debugger attempts to write trigger N with dmode=0 and chain=1

Rule 1 is not violated because the native debugger did not overwrite a trigger that had dmode=1.  Rule 2 doesn't apply since that only talks about actions of the external debugger.

If hardware allows the write in the scenario then we have a chain with mixed ownership and the buggy debugger is the native debugger, not the external debugger.  To prevent having mixed ownership when the external debugger has no bugs, when the next dmode=1 and there is a write of dmode=0, the chain must be set to 0 on this write.

Second, you are concerned about this:
> If the previous dmode=0 and chain=1, writing dmode to 1 should be ignored.

You've described that as a constraint but note that (unlike the first item) it is only a recommendation that does not necessarily constrain hardware.  I agree that this does seem to be covered by item 2 above, which probably explains why it describes what hardware should do and not what it must do.

YenHaoChen, 2022-09-21:
Thank you for the clarification.

closed 2022-09-21 by YenHaoChen