riscv/riscv-debug-spec #755: PMP interaction with debug spec

allenjbaum, 2022-09-27:
The  debug spec section A.2 says:
     Note that for debug to be possible, the PMP must not disallow fetches, loads, or stores
      in the address range associated with the Debug Module when the hart is in Debug Mode.

This is ambiguous. 
This could mean that there is a hidden  PMP entry that allows 
all accesses to the debug range when the hart is in debug mode.
Or, 
it could mean that Mm-de SW must never configure a (highest priority) locked entry that 
covers the debug module address range with anything other than RWX permissions.

Which, if any, of these interpretations is correct? 
Or are both correct; it is an implementation choice?

The converse of this is also unclear: can Mmode SW access the debug mode address range without restrictions?
 
I am assuming that, in debug mode, the PMP ), and that

timsifive, 2022-09-28:
I read it as "the PMP must not ...". The PMP is part of the hardware. It must enforce this requirement. It mustn't hope that software doesn't configure something like that.

pdonahue-ventana, 2022-09-28:
That matches with my interpretation.  It seems like the external debugger shouldn't need to rely on software (even trusted M-mode software) to be able to debug things.  Could this be clarified by saying "the PMP **hardware** must not disallow..."?

allenjbaum, 2022-09-28:
Hmm, still too ambiguous to me. IF an entry is programmed to disallow
access to some range, then it is the hardware that prevents that access
(based on its state)
A strong statement would be "the PMP *hardware* must not disallow... in
Dmode regardless of how it is configured"

closed 2022-09-29 by timsifive

allenjbaum, 2022-10-11:
> completed via #758 <https://github.com/riscv/riscv-debug-spec/pull/758>.

But should it also be required that no other mode be allowed access to that
region?
That could also make it impossible.

Note that for debug to always be possible, the PMP must not disallow
fetches,
loads, or stores in the address range associated with the Debug Module when
the
hart is in Debug Mode, regardless of how the PMP is configured.

PR #758 (2022-09-29): Clarify that PMP shouldn't prevent debug.
Fixes #755.
commit 86e748ab (2022-09-29): Clarify that PMP shouldn't prevent debug. (#758)

Fixes #755.