riscv/riscv-debug-spec #756: Trigger chain with different types

YenHaoChen, 2022-09-28:
Hello,

The spec states, "Debuggers should not terminate a chain with a trigger with a different type." I think that means the spec allows a trigger chain with different types.
[image]

Consider the configuration below, where the trigger chain consists of mcontrol and mcontrol6. The trigger index=3 has a different type from others. If we allow such a chain, this trigger chain can fire on memory access that matches all six triggers.
[image]

Consider another configuration below, where the trigger chain consists of mcontrol and icount. A difference between mcontrol6 and icount is that the icount does not have the chain field. What should be the firing condition in this configuration? Is this still a chain that fires when matching all six triggers? Or, will this configuration have two distinct firing conditions, i.e., an icount (index 3) and a chain (index 4 and 5)?
[image]

pdonahue-ventana, 2022-09-28:
> Consider the configuration below, where the trigger chain consists of mcontrol and mcontrol6. The trigger index=3 has a different type from others. If we allow such a chain, this trigger chain can fire on memory access that matches all six triggers.

Implementations are not expected to support both mcontrol and mcontrol6.  The top of the mcontrol6 section says that it "replaces mcontrol in newer implementations".  Section 1.2.1.4 says that one new feature of v1.0 is that mcontrol is deprecated.  Maybe the mcontrol section should have a more prominent statement that it is deprecated and only described in the spec for backward compatibility.

> Consider another configuration below, where the trigger chain consists of mcontrol and icount. A difference between mcontrol6 and icount is that the icount does not have the chain field. What should [...]

The chain ends with the first trigger that doesn't have chain=1.  Since icount doesn't have a chain field, it must terminate the chain.  Therefore, you have two chains.  The first one mixes three mcontrol triggers (type 2) and an icount trigger (type 3).  Because they have different types, the sentence you highlighted says that it is undefined when the chain fires.  The second chain has two mcontrol triggers and will behave as defined in the rest of the spec.

YenHaoChen, 2022-09-29:
Thank you for the clarification.

closed 2022-09-29 by YenHaoChen