riscv/riscv-debug-spec #757: Trigger chain's priority and firing time

YenHaoChen, 2022-09-28:
Hello

The spec defines the priority of a trigger chain as the lowest priority of the triggers in the chain. I think this also establishes the timing of the chain since priority is associated with timing (before/after).
[image]

However, another sentence states the timing is undefined, which seems contradictory to me.
[image]

Can you share some words about this conflict?

Thank you

pdonahue-ventana, 2022-09-28:
> The spec defines the priority of a trigger chain as the lowest priority of the triggers in the chain. I think this also establishes the timing of the chain since priority is associated with timing (before/after).

Yes.

> However, another sentence states the timing is undefined, which seems contradictory to me.

That says:
> Debuggers should not terminate a chain with a trigger with a different type. It is undefined when exactly such a chain fires.

Since you're asking about having different types and also different timings in the same chain, are you considering mixing type 2 and type 6 in the same chain (since those are the only two types that have a timing field)?  The intention is that type 2 is deprecated and that new implementations should support type 6 but not type 2.  It's unclear to me whether such an unusual chain would have undefined behavior or whether the normal (single type) behavior applies:
> A chain of triggers that don’t all have the same timing value will never fire.

Because we don't anticipate anyone supporting type 2 and type 6 in the same implementation and even if they're both available we don't expect debuggers to mix them, I'm not sure that it's worth clarifying whether the chain would never fire or if it's undefined.

YenHaoChen, 2022-09-29:
Thank you. I have a more precise concept of design attention.

closed 2022-09-29 by YenHaoChen