riscv/riscv-debug-spec #766: stoptime description needs clarification

aswaterman, 2022-10-18:
As currently written, the definition of the `stoptime` field implies that `time` and `mtime` are handled differently: that `stoptime` affects the `time` CSR but only affects `mtime` when all harts have `stoptime`=1.  But `time` and `mtime` are not independently controllable.  As the priv spec says, "The time CSR is a read-only shadow of the memory-mapped mtime register".

This might just be a matter of imprecise wording.  I suggest clarifying it by dropping all reference to `time`.  Here's a possible rewording that retains all of the semantic information that isn't contradictory:

"If `stoptime`=0, then `mtime` increments as usual.  If all harts have `stoptime`=1, then `mtime` is allowed to stop incrementing."

Of course, this does mean that `stoptime` might have no effect even when all harts have `stoptime`=1, which is odd.  But that is what the current spec effectively says, and I'm only trying to clarify, not make any normative changes.

Looks like @pdonahue-ventana most recently edited the text in 6ea272fc07b6ec7409cf46b3ab9beac44bf84e2b

cc @jhauser-us, who pointed this out.

aswaterman, 2022-10-19:
My point is that they aren’t independently controllable—that would contradict the priv spec, which says one is a shadow of the other. The debug spec can’t override that.

aswaterman, 2022-10-19:
The questions you raise are not within the bailiwick of the debug specification.  Perhaps we should resolve them elsewhere and revisit this concern later.  Since you're the one who brought it up to begin with, I'll leave it to you to decide whether to close this ticket for the time being.

jhauser-us, 2022-10-19:
Sorry for muddying this issue.  My concerns about ambiguity in the ISA specs need to be addressed elsewhere.  It came up here because I'd intended to back you up, Andrew, by quoting the ISA specs, then realized the specs didn't actually let me do that.

Getting back to the point, the intention of the specs has long been that a RISC-V system with multiple harts has a single, global `mtime`, and that the read-only `time` CSR in every hart is a copy of this global `mtime`.  (The value read from `time` may include a small time offset due to physical communication delays, but the system needs to try to equalize this offset across all harts.)  Given this intention, it's impossible to stop incrementing `mtime` for one hart separately from other harts, or to stop incrementing any hart's `time` CSR independently of `mtime`.

pdonahue-ventana, 2022-10-19:
> My point is that they aren’t independently controllable—that would contradict the priv spec, which says one is a shadow of the other. The debug spec can’t override that.

The debug spec introduces a new mode.  It cannot introduce new optional behavior that is visible only in that mode?  If the answer is no then it's unfortunate that this wasn't known back when debug 0.13 was ratified or in the intervening years when implementations were created.

jhauser-us, 2022-10-19:
First off, the normal physical implementation of `time` doesn't have a counter for it that's separate from `mtime`.

But more importantly, in a multi-hart system, when a hart is resumed (leaves debug mode), how does the debug spec propose to make `time` be the same again for all harts as the unprivileged spec intends?  Certainly the debug spec can make its own rules for debug mode, but what is the point of stopping `time` advancing while in debug mode if the effects of the stoppage are just going to be undone the moment you leave debug mode?

> it's unfortunate that this wasn't known back when debug 0.13 was ratified or in the intervening years when implementations were created.

I don't know, but it's conceivable to me that all actual debug modules for multi-hart systems have the `stoptime` bit hardwired to 0, given that there's no way to support it consistent with the ISA specs.

pdonahue-ventana, 2022-10-19:
> I don't know, but it's conceivable to me that all actual debug modules for multi-hart systems have the stoptime bit hardwired to 0, given that there's no way to support it consistent with the ISA specs.

I honestly don't understand how it's not consistent with the ISA specs*.  The debug spec modifies the time CSR behavior only in debug mode, not in any way that's visible to any mode that is specified in the privileged spec.  The debug spec modifies all sorts of other things in debug mode: mstatus.MPRV behavior, NMIs are masked, branches and auipc may be illegal instructions, etc.  Is it a problem to do these things in debug mode?  In what way are these substantively different than allowing the time CSR to have this unusual behavior in debug mode?

I don't want to give the impression that I love stoptime.  (Quite the opposite.)  My concern is solely with backward-compatibility.

\* I'm assuming that you mean ISA specs other than this CSR definition in the ratified debug spec which, since it's a CSR, is arguably an ISA spec.

aswaterman, 2022-10-19:
If the intent really was to contravene the statement in the priv spec that time is a shadow of mtime, then it should’ve been stated more explicitly. I’m sure the reason this escaped notice in review is that it wasn’t at all obvious it was introducing a new mode with material cost and non-obvious benefit. In any case, practical multicore systems are sure to hardwire this bit to 0 anyway, so perhaps the feature being mis-specified is moot.

pdonahue-ventana, 2022-10-19:
> material cost and non-obvious benefit

I wouldn't call my example's clock gater on some flops a material cost.  But I agree that there's non-obvious benefit (my theoretical example notwithstanding).

So should we make your proposed change and add that to the list of incompatible changes in the introduction?  Or should we stick with what it says now?  Or keep the current text but add some warnings about this (other than [the sentence](https://github.com/riscv/riscv-debug-spec/blob/master/Sdext.tex#L35) in section 4.1 that time "will resynchronize with mtime after leaving Debug Mode")?  Or something else?

aswaterman, 2022-10-19:
I don't feel comfortable making a unilateral decision.  My preference is still to make my proposed change, but that requires buy-in from others.

pdonahue-ventana, 2022-10-19:
Sure.  Different implementations always have different costs.  In the SiFive implementation, stoptime would be painful and they would probably hardwire stoptime=0.  In my example implementation, there's a more complicated time distribution network (which presumably exists for other reasons and was not put in place just for this obscure feature) but it's quite easy to implement stoptime as specified.

I did some archaeology.  This bit was [in the very first commit to the debug spec in January 2017](https://github.com/riscv/riscv-debug-spec/commit/ceb5d66a7e0cb5cea0511b50849ee4acc7baaa01#diff-f17f864353ec1b30c335efa35d3bf47a3b45dad4ff5be736da8176c86b5e0a45R40) as:
> Controls the behavior of any timers while the component is in Debug Mode.  This includes the {\tt time} and {tt timeh} CSRs.  When 1, timers are stopped when the component is in Halt Mode.  Otherwise, the timers continue to run.
>
> An implementation may choose not to support writing to this bit. The debugger must read back the value it writes to check whether the feature is supported.

It's interesting that complying with this would be difficult in a SiFive implementation since the [sole author of the initial draft](https://github.com/riscv/riscv-debug-spec/commit/ceb5d66a7e0cb5cea0511b50849ee4acc7baaa01#diff-c42542cffe53098196ac1a9238975111cf4af5430f9dc5a1dd2c647ca0a2a814R91) was from SiFive.  Perhaps it was added for someone else and SiFive always intended to take advantage of the fact that this was optional.

timsifive, 2022-10-21:
From my recollection, stoptime was added so that a hart can't tell that it was temporarily halted. This is useful when debugging code that has a regular timer interrupt configured. Debugging typically takes seconds, and a timer interrupts usually fire more often than that. So if you just want to step through your code without ending up in interrupt handlers all the time, having time frozen when the hart is halted is a nice feature. This makes a lot of sense in a single hart system. In a multi-hart system where you're not even halting every hart, it becomes more dubious.

> Getting back to the point, the intention of the specs has long been that a RISC-V system with multiple harts has a single, global mtime, and that the read-only time CSR in every hart is a copy of this global mtime.

If time is simply a shadow of mtime, then stopping time doesn't affect the code being debugged in any way.

There could be a use for that as well, though. It lets the debugger know the value of time when the halt occurred. This would be nice to know, e.g. to see if an interrupt didn't happen yet because it's time didn't come, or perhaps something else was misconfigured.

To allow all this, I think we should change the definition of stoptime to reference whether `time` is reading the value of `mtime`, or the value `mtime` had when Debug Mode was entered. We can leave the language that allows `mtime` to stop incrementing if all harts have `stoptime=1` and are in Debug Mode.

aswaterman, 2022-10-26:
We discussed this in the AR meeting, and concluded that, since the feature is ratified, it would be better to avoid any backwards incompatible changes--which is harmless, since the feature is optional.  We hadn't discussed Tim's most recent suggestion of broadening the definition to allow either behavior (`time` is allowed to either stop or continue shadowing `mtime`).  IMO, that is a reasonable route to go, because it avoids invalidating existing implementations.

Alternatively, if we go the "do-nothing" route, we suggest adding some non-normative text that explains the semantics of stopping `time` while continuing to increment `mtime` (namely, that as soon as debug mode is exited, `time` must snap back to shadowing `mtime`).  We also suggest adding a note that we expect the bit to be hardwired to 0 in many multi-hart implementations.

Regardless, we think the debug TG should propose how to proceed.

pdonahue-ventana, 2022-10-26:
> Alternatively, if we go the "do-nothing" route, we suggest adding some non-normative text that explains the semantics of stopping time while continuing to increment mtime (namely, that as soon as debug mode is exited, time must snap back to shadowing mtime).

[Section 4.1](https://github.com/riscv/riscv-debug-spec/blob/master/Sdext.tex#L35) already says:
> If stoptime is 0 then time continues to update. If it is 1 then time will not update. It will resynchronize with mtime after leaving Debug Mode.

(I thought that I had already mentioned that in this issue but I guess not.)

As for the bigger question, we can ask the TG and come up with a path forward.

aswaterman, 2022-10-26:
> (I thought that I had already mentioned that in this issue but I guess not.)

Sorry if I missed it.  Nevermind that aspect, then.

pdonahue-ventana, 2023-03-28:
There were two items:
- adding some non-normative text that explains the semantics of stopping time while continuing to increment mtime (namely, that as soon as debug mode is exited, time must snap back to shadowing mtime).
- adding a note that we expect the bit to be hardwired to 0 in many multi-hart implementations

The first one was already in the spec (as mentioned in my comment above).  The second one was addressed by #769.  I think that we can retire this one.

closed 2023-03-28 by pdonahue-ventana