riscv/riscv-debug-spec #771: PMP interaction with debug spec, updated text

silabs-robin, 2022-10-28:
Hi.

Referring to the changes made in this https://github.com/riscv/riscv-debug-spec/pull/758 PR, relating to this https://github.com/riscv/riscv-debug-spec/issues/755 issue.

> Note that for debug to always be possible, the PMP must not disallow fetches, loads, or stores in the address range associated with the Debug Module when the hart is in Debug Mode, regardless of how the PMP is configured.

There are several potential issues with that sentence, as detailed below.

--------

The sentence is still ambiguous.

It could mean:
_"If an implementation requires debug to always be possible, then [..]"_
Or:
_"Debug must always be possible, therefore [...]"_

_(I could not find anywhere in the spec stating that debug must always be possible.)_

Ambiguities are important to clean up. I had, for instance, interpreted the previous text as Allen's second alternative (SW) before that ambiguity was now resolved.

--------

We are worried that overruling the PMP settings is a potential security hole, and therefore prefer if the sentence is changed to explicitly make both interpretations a legal option (i.e. overruling the PMP or not overruling the PMP).

Suppose that a debug-related signal is glitched high, then the PMP settings become overruled and we would loose the extra protection that could have been provided by the PMP.
For example, software could configure the PMP such that debug access is never allowed in the field; even if a debug-related signal is glitched the PMP would prevent debug mode execution.

--------

Lastly, should this also include the PMA and not only the PMP?

--------

ps.

I know this is about an appendix and not "the spec itself", but it is bundled with the spec and people use this as  a reference and it would be good to avoid confusion.

Regards Robin.

pdonahue-ventana, 2022-11-11:
> (I could not find anywhere in the spec stating that debug must always be possible.)

Debug must always be possible because the spec says that if you do X then Y happens, not that Y happens only if debug is possible. (e.g. an abstract command with transfer=1 and write=0 will read the register, not that it will read the register sometimes but it will be ignored other times)

> We are worried that overruling the PMP settings is a potential security hole, and therefore prefer if the sentence is changed to explicitly make both interpretations a legal option (i.e. overruling the PMP or not overruling the PMP).

You propose to allow two options:
1. Accesses made from debug mode to the debug module bypass PMP checks.
2. Accesses made from debug mode to the debug module honor the PMP settings put in place by M mode.

Option 2 might allow debug mode to access the debug module even though M mode itself cannot.  I don't see a security issue since the code running in debug mode is either the park loop which is code that comes from the debug module itself (and must be inherently trustworthy) or it comes from the program buffer which was generated by the external debugger which has full access to the debug module already.

If M mode software was able to prevent debug mode from accessing the debug module then the hart will get into an infinite loop of access faults which itself is a security problem (denial-of-service).  If resethaltreq=1 then this hang can't even be solved by resetting the hart (only by resetting the hart and the DM).  There's an authentication mechanism to prevent untrusted external debuggers from causing entry to debug mode.  If someone is able to get into debug mode then they are inherently trusted on RISC-V.  (Note that this is somewhat different than the ARM architecture where there are untrusted external debuggers which can only see non-secure state and trusted external debuggers which can see everything.)

> Suppose that a debug-related signal is glitched high, then the PMP settings become overruled and we would loose the extra protection that could have been provided by the PMP.

This only affects accesses while in debug mode.  If we entered debug mode due to such a glitch (presumably on haltreq due to a buggy DM implementation) and there is no actual external debugger then both options 1 and 2 will hang.  Option 2 will get in an infinite loop of traps while option 1 will get in an infinite loop of loads from an address in the DM.  That load will not access secret data.  It will just be the value that indicates that the DM has no commands for the halted hart to process.

M mode shouldn't be able to prevent a higher mode (which debug mode essentially is) from accessing resources that are necessary for that mode to function.  M mode cannot prevent debug mode from accessing dcsr, dpc, dscratch and it should not be able to prevent debug mode from accessing the DM.  M mode can, however, prevent debug mode (and M mode itself) from accessing anything else in memory which is not fundamentally necessary for debug mode to be able to access.

> Lastly, should this also include the PMA and not only the PMP?

Yes.

I'll make a PR that clarifies the ambiguity you pointed out and that mentions PMA.

silabs-robin, 2022-11-22:
Related PR closed.

closed 2022-11-22 by silabs-robin