riscv/riscv-debug-spec #772: Debug CSR DCSR/DPC/DSCRATCH access using CSR instructions

jrahmeh, 2022-10-28:
Section 4.8 of the debug spec (version 1.0.0) states that:

> They are CSRs, accessible using the RISC-V csr opcodes and optionally also using abstract debug commands.

However, the RISCV privileged spec states that:

> Machine-mode standard read-write CSRs 0x7A0–0x7BF are reserved for use by the debug system. Of these CSRs, 0x7A0–0x7AF are accessible to machine mode, whereas 0x7B0–0x7BF are only visible to debug mode. Implementations should raise illegal instruction exceptions on machine-mode access to the latter set of registers.

If a hart is executing a csr opcode it must be either running (not in debug mode) or executing from the program buffer.  In either case, the privilege mode is M or lower (section 4.1 of debug spec says program buffer executes in M privilege).  So is DCSR  accessible using a csr opcode?

pdonahue-ventana, 2022-10-28:
> If a hart is executing a csr opcode it must be either running (not in debug mode) or executing from the program buffer. In either case, the privilege mode is M or lower (section 4.1 of debug spec says program buffer executes in M privilege).

You're always in Debug Mode when you're halted.  For instance, section 4.1 says "Debug Mode is a special processor mode used only when a hart is halted for external debugging" and "When executing code due to an abstract command, the hart stays in Debug Mode."  Debug Mode is a distinct mode.

Section 4.8 says "These registers are only accessible from Debug Mode."  And the privileged spec agrees that those CSR addresses are "only visible to debug mode."

However, I see the confusion.  The intention of "All operations are executed with machine mode privilege" is that memory operations behave as if they're in M mode (using bare physical addresses) and that all CSRs accessible to M mode are accessible to Debug mode.

jrahmeh, 2022-10-28:
Would it be possible to add that to the spec to make it clearer? Maybe something like: "All operations are executed in debug mode which is identical to M mode for memory access and has additional privileges than M mode for CSR access."

pdonahue-ventana, 2022-10-28:
Yes.  I'll do something like that when I get a chance.  This can remain open in the meantime.

pdonahue-ventana, 2023-03-21:
Fixed in #773.

closed 2023-03-21 by pdonahue-ventana

pdonahue-ventana, 2024-04-04:
Yes.  These Sdext CSRs are listed in section 4.10 which says "These registers are only accessible from Debug Mode."