riscv/riscv-debug-spec #775: Do Etrigger and Itrigger provide functionality beyond that of a PC trigger?

jrahmeh, 2022-11-15:
Given that a PC trigger (sel=0, exec=1, match=0) can be used to stop (generate a breakpoint exception or enter debug mode) before the first instruction of a trap handler, what additional functionality is provided by etrigger/itrigger?  Why not simplify the spec by removing etrigger/itrigger or placing them in an appendix?

pdonahue-ventana, 2022-11-15:
etrigger can filter on the mode that caused the exception and on specific exception types.  PC triggers are only able to stop when you enter a particular handler.  All synchronous exceptions go to the same vector and, unless you're in Vectored mode, all asynchronous interrupts go there, too.  If you're debugging illegal instructions in U mode, it's nicer to only stop if you actually hit an illegal instruction in U mode.  With the PC trigger on the vector, you'd potentially stop every time you got an interrupt, page fault, misaligned load, ecall, etc.

It's true that you could set a PC trigger at a subhandler that runs after mcause has been parsed, though setting such a trigger would require you to understand the initial handler code and trust it to be bug-free.

jrahmeh, 2022-11-15:
Thanks for the clarification.  Closing issue.

closed 2022-11-15 by jrahmeh