riscv/riscv-debug-spec #778: Clarification on DPC bit 0

tovine, 2022-11-29:
I have a question about DPC, and more specifically bit 0 of this register.
In the spec it's listed as a DXLEN register, R/W all the way down to bit 0 while the _PC_ is not (at least it's impossible to have bit 0 of the PC set to 1 by any legal instruction).
[image]

What is the expected behavior when writing a value to DPC where bit 0 is set, and what should then be the result of resuming execution in such a case?
I would assume that DPC should follow the PC behavior and in reality only be defined for [DXLEN-1:1] with bit 0 being hardwired 0, but what's your opinion, and could we get a clarification in the spec on this one?
The only other valid alternatives as I see it would be:
a) Bit 0 of DPC is ignored when resuming, so PC is set to DPC[DXLEN-1:1]
b) The hart raises a Instruction address misaligned exception (even if the C extension is implemented, which contradicts the note in the C spec that says this exception type is impossible when compressed instructions are supported)

pdonahue-ventana, 2022-11-29:
It should be the same as mepc/sepc/vsepc where the low bit is read-only zero (or the low two bits if the C extension isn't implemented).  I made a clarification in #779.

tovine, 2022-12-22:
Thanks, new spec looks clear. 🙂

closed 2022-12-22 by tovine