riscv/riscv-debug-spec #783: Action 8 - 9

Silabs-ArjanB, 2022-12-07:
In 'table 5.1 action encoding' the values 8 - 9 are described as follows:

> Signal the firing of the trigger to other blocks within the hart (e.g. as countable events to hpmcounters). Use external debug trigger output 0 or 1 (respectively).

Why does it specifically state 'within the hart'? That seems to conflict with 'use external debug trigger output 0 or 1' as external triggers are 'abstract concepts that can signal the DM and/or receive signals from the DM' (i.e. there is no notion of 'within the hart' in respect to external triggers.

Could the description simply be as follows?

> Signal the firing of the trigger to other blocks via external debug trigger output 0 or 1 (respectively).

pdonahue-ventana, 2022-12-07:
tmexttrigger is a trigger that's external to the TM and that is defined to connect to something within the hart.  dmexttrigger is a trigger that's external to the DM and is intended to connect to something else in the system such as another DM.

An example of the former is given in the text.  You might want to have HPM count how many times you've executed a particular PC, loaded from a particular address, etc.

An example of the latter is if you have multiple DMs in the system and you want the benefit of halt groups and resume groups.  With a single DM, you can have the halting of a single hart cause other harts to halt.  With multiple DMs, you can still have the single halting hart use halt groups to halt other harts (if any) controlled by the same DM but you can simultaneously use dmexttrigger to have that DM send a trigger to other DMs which can then be programmed to halt a halt group.  This inter-DM cross triggering allows all harts in the system to halt.

The spec could theoretically allow TM external trigger outputs to connect to DM external trigger inputs.  I don't think that's useful since there's nothing you can do with these DM inputs other than halt/resume groups.  The only reason I can see to connect things like this is to have the fact that hart A executed a particular PC cause harts B and C to halt but not cause hart A to halt (by having B and C but not A in a halt group which is triggered by the input).  If you want to halt hart A then you'd just use action=1 (and optionally add B and C to a halt group if you want them to also halt).  This seems like a strange thing to do but I may be missing something.

Silabs-ArjanB, 2022-12-08:
Hello @pdonahue-ventana Thank you for your answer. Let me give an example of what kind of functionality I am looking for. Suppose I want to toggle some chip general purpose IO pin when the hart is accessing a certain address or executing an instruction at a specific address. Specifically we do not want to enter debug mode when this happens as that is considered too intrusive.

Do I though I could do that as follows:

- Usage of an mcontrol6 trigger to trigger at a specific PC, load/store address, etc.
- mcontrol6 would be set up with action = 8

Upon a mcontrol6 trigger match it is my interpretation from table 5.1 that a the match would be signaled on _debug module external trigger_ output 0 (which I could then convert into toggling of some IO pin at chip level).

So my question is unrelated to tmexttrigger (as I would be using a trigger with type=6 in above example). 

Table 5.1 talks about 'use external debug trigger output 0 or 1'. I assume that means 'use **debug module external trigger** output 0 or 1'; is that the correct interpretation? If so, then it still seems conflicting to me to use 'within the hart' and 'debug module external trigger' in the same paragraph.

timsifive, 2023-02-20:
Using mcontrol6 to signal something outside the hart sounds useful, and I see no reason to disallow it.

The text is confusing, referring to both "within the hart" and "external debug trigger." It's possible to implement Sdext even without a Debug Module, and even in that case it would make sense to allow these actions to send some signal outside the hart or even off-chip.

I hope to get back to this to clarify the text, but feel free to make a suggestion or PR.

closed 2023-04-11 by timsifive