riscv/riscv-debug-spec #785: Serial Wire Debug (SWD)

stayforapple, 2022-12-09:
Serial Wire Debug (SWD) is a 2-pin (SWDIO/SWCLK) electrical alternative JTAG interface that has the same JTAG protocol on top.

It is highly recommended to use the SWD mode when the size of the circuit board is limited. It requires fewer pins, and of course the required PCB space is small! 

https://developer.arm.com/documentation/ihi0031/a/The-Serial-Wire-Debug-Port--SW-DP-/Introduction-to-the-ARM-Serial-Wire-Debug--SWD--protocol?lang=en

Does RISC-V support such powerful and popular Debug interface ? I found nothing related or similar in the Version 1.0.0-STABLE.

timsifive, 2022-12-09:
Like you said, there is nothing in the spec right now. The simplest way to get to two pins right now is to use cJTAG. That's not explicitly mentioned in the spec either, but there are standard ways to transport JTAG over cJTAG.

I'm all for defining a SWD DTM in the debug spec. The best way to get that started is probably to write up a proposal and discuss it in a PR here and/or on the mailing list.

stayforapple, 2022-12-10:
Thanks, Tim. We do need the Serial Wire Debug or the cJTAG or the similar interface, which should not be absent in the RISC-V debug spec.

How to write up a proposal ? Is there any template ? I have never joined the development of any microprocessor before.

timsifive, 2022-12-12:
> How to write up a proposal ? Is there any template ? I have never joined the development of any microprocessor before.

Don't overthink it. Write a description that technically savvy people will be able to understand. I find that you get better feedback if you propose something than if you simply ask people what they think. Essentially what you're defining is the behavior of a black box which has SWD pins on one end, and connects to the Debug Module Interface on the other. (See the System Overview figure in the spec if that doesn't make sense.)

Eventually we'll want a chapter similar to 6.1 (JTAG DTM) for a cJTAG DTM. (There might be some RISC-V Foundation process to deal with as well, but don't let this get in the way of starting the actual work that needs to be done.)

stayforapple, 2022-12-16:
Thanks, Tim, let me think it over.

What's the difference between IEEE-defined cJTAG and ARM-defined SWD ?

Both are the two-pin connection, one pin is for the clock and the other for the bidirectional data flow.

pdonahue-ventana, 2022-12-16:
Before adding anything about SWD to the spec, please read the [RISC-V Encumbered Information policy](https://docs.google.com/document/d/1iiRl0eSCcZFTmTp370CPX-NasdGDO1bQF4m5k8YKq-o/edit).

tovine, 2022-12-21:
Just FYI: even though it's not mentioned in the RISC-V debug spec, several (if not most) debug probe vendors already support debugging RISC-V over SWD.
The way the debug spec is currently written (or at least interpreted) it doesn't really care what the physical interface/transport layer is as long as it can access the DM registers at the provided addresses. 🙂

bruceable, 2022-12-21:
Can you name a RISC-V chip/SoC vendor that supports SWD?  I don't know that
much about Arm's SWD protocol but it isn't the same as cJTAG, correct?  At
SiFive, we only support JTAG and cJTAG (4-wire, 2-wire).

--- Bruce Ableidinger, SiFive, Inc.

tovine, 2022-12-21:
It is not the same as cJTAG, no.
After a quick look through the hardware list at the RISC-V Exchange I could find this (a Chinese one, CH32V2x):
[image]
Also it seems very much like the RV32M1 from NXP uses it: https://open-isa.org
[image]

stayforapple, 2022-12-22:
Great, how can they realize the SWD ? Is there any open-source protocol ?

The SWD interface should be incorporated into the RISC-V specification as soon as possible !

The compatibility with ARM will boost the popularity of RISC-V !

tovine, 2022-12-22:
I'm not sure it's that simple; while I'm not a lawyer, I suspect that the Arm Serial Wire Debug protocol and [specification](https://developer.arm.com/documentation/ihi0031/a/The-Serial-Wire-Debug-Port--SW-DP-) is Arm proprietary information and as such is probably unsuitable for inclusion in a RISC-V spec.

In most (if not all) cases, SoC vendors include the SWD access port as part of a debug subsystem licensed from Arm - either through a Cortex-M bundle or a CoreSight SoC package. The latter would probably be most likely in case of a RISC-V core in such a system.
I haven't heard about any open source implementation, or whether such an implementation would even be permitted to exist without legal issues.

stayforapple, 2022-12-22:
Thanks a lot. If the popular SWD cannot be included in the RISC-V specification, the cJTAG seems to be the only option then.

What's the difference between IEEE-defined cJTAG and ARM-defined SWD, anyway ? Which has the superior performance ?

I am just curious, as a fully-matured architecture, ARM developed its own SWD instead of following the IEEE-defined cJTAG, why ?

tovine, 2022-12-22:
I haven't looked at comparing the SWD and cJTAG side by side so I don't know what the real differences are (but I suspect the protocols are different, since both have a clock and a data line).
But since cJTAG (part of the IEEE 1149.7 standard) was only ratified in 2009 (and released in 2010) it wasn't available when Arm developed SWD (current version of the spec documents I've seen was from 2006, not sure if the standard is even older than that)...

stayforapple, 2022-12-22:
Thanks. I just found the following information,

The ARM SWD technology, introduced in 2004, is widely deployed in silicon and supported by a large number of tools vendors. SWD replaces the alternative 5-pin JTAG debug port providing all the normal debug functionality with just two pins, increasing debug performance for ARM Cortex™ processors, and avoiding the need to use scan chains for debug. 

The new multi-drop SWD is fully backwards compatible, retaining existing single point-to-point host equipment connections, and enables a device to power down completely while that device is not selected, reducing power consumption.  The ARM multi-drop SWD has been designed to be fully interoperable with the recently announced IEEE 1149.7 standard.

https://www.design-reuse.com/news/21605/serial-wire-debug-swd.html

tovine, 2025-10-23:
Now that it's public info, I can say that the Nordic Semiconductor nRF54 series SoCs (like the [nRF54L15](https://www.nordicsemi.com/Products/nRF54L15)) also have RISC-V debugging via SWD (supported by among others Segger J-Link and Ashling debug tools).