riscv/riscv-debug-spec #790: hit bit descriptions

pdonahue-ventana, 2022-12-22:
The description of the hit bit in mcontrol and mcontrol6 says:
> If this bit is implemented then it must become set when this trigger fires and may become set when this trigger matches.

The description of hit in icount (where matching and firing are definitely distinct) says:
> If this bit is implemented, the hardware sets it when this trigger fires. 

The description in itrigger, etrigger, and tmexttrigger says:
> If this bit is implemented, the hardware sets it when this trigger matches. 

I question the use of "matches" on the last three.  It's not immediately clear to me that any of them can match without firing.  If match implies fire then the word doesn't matter except that using consistent language ("fire") would be better.  If match does not imply fire then we probably want it to be set only when firing (like on icount where matching and firing are more decoupled).  Either way, I think those three should refer to firing.

What do you think?

timsifive, 2022-12-27:
mcontrol/mcontrol6 are allowed to set on match/fire to make hardware's life simpler when dealing with chains.

The same argument could be made for itrigger, etrigger, and tmexttrigger. They could match but not fire if another trigger matches+fires at the same time. The spec says:
> When multiple triggers in the same priority fire at once, hit (if implemented) is set for all of them.
> If more than one of these triggers has action =0 then tval is updated in accordance with one of
> them, but which one is unspecified.

That seems consistent with hit being set on match, not on fire.

timsifive, 2023-03-21:
> How can etrigger (highest priority) match without firing?

I don't think it can. Only icount and mcontrol* (due to chaining) can match without firing.

Cleaner language sounds good, but the devil is in the details. Do you want a PR where it's easier to discuss actual wording?

pdonahue-ventana, 2023-03-28:
This is not very important to me.  I'll try to propose a clarification in a PR.  If that doesn't happen then no big deal.

pdonahue-ventana, 2023-04-18:
This is not that important to me.

closed 2023-04-18 by pdonahue-ventana