riscv/riscv-debug-spec #792: Question about Table 5.10 Suggested Trigger Timings

YenHaoChen, 2023-01-04:
Table 5.10 gives suggested trigger timings. I am confused about the terminology "Address+Data" ("Address+Instruction"). The "Address" and "Data" ("Instruction") are controlled by the same bit "select" and can only be either one. What does the "+" sign mean? Does it mean "or"?
```
Match type                  | Encoding                               | Suggested trigger timing
-----------------------------------------------------------------------------------------------
execute address             | select=0, {execute,store,load}={1,0,0} | before
execute instruction         | select=1, {execute,store,load}={1,0,0} | before
execute address+instruction | select=?, {execute,store,load}={1,0,0} | before
load address                | select=0, {execute,store,load}={0,0,1} | before
load data                   | select=1, {execute,store,load}={0,0,1} | after
load address+data           | select=?, {execute,store,load}={0,0,1} | after
store address               | select=0, {execute,store,load}={0,1,0} | before
store data                  | select=1, {execute,store,load}={0,1,0} | before
store address+data          | select=?, {execute,store,load}={0,1,0} | before
```

[image]

scottj97, 2023-01-05:
Perhaps the address+data is referring to a chain of triggers? Below Table 5.10 is this line:

> A chain of triggers that don’t all have the same timing value will never fire. That means to implement the suggestions in Table 5.10, both timings should be supported on load address triggers.

pdonahue-ventana, 2023-01-05:
Yes, this has to do with chaining.  If you only trigger on a load address, use timing=before.  If you only trigger on a load data value, use timing=after.  If you have a chain that fires only when you load particular data value from a particular address then use timing=after.

YenHaoChen, 2023-01-05:
Thanks for the reply. Do we have suggested timing for ```select=1(data), {execute,store,load}={1,0,1}```, ```select=1(data), {execute,store,load}={0,1,1}```, and ```select=1(data), {execute,store,load}={1,1,1}```?

pdonahue-ventana, 2023-01-06:
I think that there is no suggested timing for those since there is no single timing that provides "the best user experience."

closed 2023-01-06 by YenHaoChen