riscv/riscv-debug-spec #794: Is it permissible to hardwire dcsr.stepie=1 ?

JanMatCodasip, 2023-01-18:
Q1)
Is it allowed to hard-wire `dcsr.stepie` to value `1` (that is, interrupts always enabled during single-step)?

The description of dcsr.stepie in the spec allows hard-wiring to 0 but not mentions hard-wiring to 1.
At the same time, dcsr.stepie is marked as WARL.
Shall this be interpreted that hardwiring dcsr.stepie to 1 is permitted or forbidden?

Q2)
What is the preference of the debug tools when it comes for interrupts during stepping - enabled or disabled? (In case anyone here is able to shed more light on what debug tools do or expect.)

Thank you!

pdonahue-ventana, 2023-01-18:
See https://lists.riscv.org/g/tech-debug/topic/81443579 for a discussion of this.  In particular, this comment from someone at IAR (a debugger company):
> It is better to NEVER step into interrupts then ALWAYS step into them, so current reset default=0 is better.
Debugger must set this bit to '1' if stepping into interrupts is desired (which is rarely the case ...).

JanMatCodasip, 2023-01-19:
Thank you for your reply.

In that case, would it be worth to add a note to the spec that support for dcsr.stepie=0 is strongly recommended for best debug experience (to undesired interrupts during an interactive debug session)?

JanMatCodasip, 2023-01-19:
Or perhaps even a stronger requirement: `dcsr.stepie=0 shall be supported` ?

timsifive, 2023-01-23:
I think stepie=0 should be supported. Requiring it would not be backwards compatible, and this doesn't feel important enough for that kind of change.

JanMatCodasip, 2023-02-02:
Thank you for replying. 

In another words, do I read it correctly that: `"support for stepie=0 is recommended but not required"` ?

If that is correct understanding, do you think it is worth to add that recommendation to the spec?

pdonahue-ventana, 2023-03-21:
Fixed in #808.

closed 2023-03-21 by pdonahue-ventana