riscv/riscv-debug-spec #796: Priority of previous-instruction debug exceptions and interrupts

JamesKenneyImperas, 2023-01-26:
Section 5.3 (table 5.2) indicates the priority of synchronous exceptions with respect to trigger-initiated debug exceptions.

Should the priority with respect to _pending interrupts_ also be defined? For example, if there is a pending trigger-initiated debug exception after the previous instruction (the first row of table 5.2) _and_ a pending enabled interrupt, which is taken? A specific case of this is when the just-executed instruction has enabled interrupts.

Thanks.

timsifive, 2023-01-31:
I think pending interrupts should be below debug interrupts. But it doesn't really matter since the timing of external interrupts is such that they don't come in when a particular instruction is being executed.

If you feel this really matters, can you make a PR that makes the spec more complete/readable?

JamesKenneyImperas, 2023-02-01:
Hi Tim,

There is this statement in the Privileged Specification about interrupts (section 3.1.9):

> These conditions for an interrupt trap to occur must be evaluated in a bounded amount of time from when an interrupt becomes, or ceases to be, pending in mip, **and must also be evaluated immediately following the execution of an xRET instruction or an explicit write to a CSR on which these interrupt trap conditions expressly depend (including mip, mie, mstatus, and mideleg)**.

This means that already-pending interrupts are treated as _synchronous_ traps when enabled because of an explicit CSR change of some kind, so I think it is worth including them in priority tables. I also feel that they should be lower priority than debug interrupts. I'll have a go at making a pull request.

James.

timsifive, 2023-02-01:
I missed that text. It's too bad they didn't include external interrupts it in the priority table in the same spec. Thanks for taking this on.

JamesKenneyImperas, 2023-02-02:
Thinking further before attempting any documentation update: the statement above implies interrupts are similar to debug events on the previous instruction with timing "after". In the absence of the debugger, the interrupt will be taken without performing any actions associated with the next instruction. Specifically, interrupts have higher priority than any page, permission, or illegal instruction fault on the next instruction.

I think this means they must be handled _after_ the "mcontrol/mcontrol6 after" stage in table 5.2 but _before_ any following stages. If the debugger is active, I think a step instruction breakpoint on the just-completed instruction should have higher priority than an interrupt on the next instruction. The various debug halt signals (`haltreq` etc) should also be higher priority than an interrupt.