riscv/riscv-debug-spec #799: Can a hart be part of multiple halt groups and if so how can this be read out?

Silabs-ArjanB, 2023-02-03:
Can a hart be part of multiple halt groups?

I would assume that the answer to that is 'yes'.

If so, is there a way that the debugger can figure out to what halt groups a hart belongs _after_ a hart has been added to its halt group(s) (other than trying to add a hart to a specific halt group while dmcs2.hgwrite == 1 and checking if dmcs2.group supported the written value and then remembering this)?

I only ask this question to check that I am not missing something here. I expect the answer will be that 'a hart can be part of multiple halt groups, but there is no way for a debugger to figure out to which halt groups a hart belongs other than remembering to which halt groups a hart was successfully added'. If that is the answer, then that's okay (I am not asking for a change; just trying to confirm if I read the spec correctly).

pdonahue-ventana, 2023-02-09:
I think that the answer is no.  There's a single group field in dmcs2 which shows "the group" that the hart is in.  Debuggers are supposed to read this back to make sure that the (singular) group is a legal group.  And the last sentence of the section says that "it must be possible to discover groups by using dmcs2" which means that the single dmcs2.group field is sufficient to express all the group information (meaning that there's a single group).

Ignoring the discovery problem, imagine that a hart can be in multiple groups.  If a hart is in group X and the debugger adds it to group Y then it would be in both X and Y.  But how could you ever remove it from group X?  It really only makes sense if adding it to group Y also removes it from group X.  Otherwise, everything would always be in group 0 even if you wanted to move it to another group.  And it would be difficult to have groups configured in one way and then later reconfigure them to another way.

Silabs-ArjanB, 2023-02-09:
Hello @pdonahue-ventana , thank you for your answer. It would then make sense that the same restriction applies to external triggers as well, right (i.e. an external trigger can only be part of 1 halt group)?

The reason I asked my original question is that I had the following scenario in mind:

- Scenario for 2 harts and 3 external input-only triggers
- Halt hart 0 when external input-only trigger 8 fires.
- Halt hart 1 when external input-only trigger 9 fires.
- Halt both harts when external input-only trigger 10 fires.
- Harts should not halt each other.

And I wanted to configure two halt groups as follows to achieve that:

- Halt group 1: Hart0 + trigger 8 + trigger 10
- Halt group 2: Hart1 + trigger 9 + trigger 10

If harts and external triggers can only be part of one halt group, then I would have to do something like the following:

- Halt group 1: Hart0 + trigger 8 (and I add logic external to the Debug Module that makes trigger 8 fire also when my original trigger 10 would fire)
- Halt group 2: Hart1 + trigger 9  (and I add logic external to the Debug Module that makes trigger 8 fire also when my original trigger 10 would fire)

I am fine with harts and triggers being part of exactly 1 halt group and exactly 1 resume group. I do however think that it would be good to make this more clear in the RISC-V Debug specification (e.g. by adding your remark that 'adding it to group Y also removes it from group X').

Best regards,
Arjan

pdonahue-ventana, 2023-02-16:
> I am fine with harts and triggers being part of exactly 1 halt group and exactly 1 resume group. I do however think that it would be good to make this more clear in the RISC-V Debug specification (e.g. by adding your remark that 'adding it to group Y also removes it from group X').

@timsifive took care of this in #804.  Not the same wording but makes the one-at-a-time concept clear.

closed 2023-02-16 by pdonahue-ventana