riscv/riscv-debug-spec #803: Can abstract commands size be grater that XLEN

hananyasegal1, 2023-02-15:
Hi,

I want to ask if according to debug spec, is it possible that abstract command may be with 64 bits size and the XLEN can be 32 bits for example?
Does it depend on debug spec version?

Thanks,
Hananya

timsifive, 2023-02-15:
The abstract command register (`command`) is always 32 bits wide, regardless of the harts the Debug Module is attached to. I don't think this was ever any different.

pdonahue-ventana, 2023-02-15:
I think that the question is whether you can have aarsize=3 (64-bit) when XLEN=32.  I believe that this would be supported on RV32D implementations.

pdonahue-ventana, 2023-03-28:
Nothing in the aarsize section says anything about XLEN.  It's all about the size of the register being accessed.  If you think that the spec is misleading or confusing then please reopen.

closed 2023-03-28 by pdonahue-ventana