riscv/riscv-debug-spec #812: Should icount fire when textra{32,64} don't match?

timsifive, 2023-03-02:
https://github.com/riscv-software-src/riscv-isa-sim/commit/d69a788be7feea8f529cb2dab0e9854f665055f5 says:

> Debug spec says about icount: "When pending is set, the trigger fires
> just before any further instructions are executed in a mode where the
> trigger is enabled"
> 
> And about icount textra: "If textra32 or textra64 are implemented for
> this trigger, it only matches when the conditions set there are
> satisfied."
> 
> I interpret that to mean that count should only be decremented when
> textra matches, but once pending, it should fire even if textra
> doesn't match anymore.

That is what the spec says, but I'm not sure it's what we want.

pending exists so that when you step an instruction which decrements icount to 0, that you then stop in the mode where the trigger is actually enabled instead of e.g. at an interrupt handler that happens to be entered then. textra.sselect and textra.mhselect do the same thing for supervisor/hypervisor context as the U, M, S, etc. bits do for execution modes. So pending should behave the same way.

Is that right?

pdonahue-ventana, 2023-04-01:
This will be fixed when #826 is merged.

timsifive, 2023-04-05:
#826 did merge.

pdonahue-ventana, 2023-04-17:
Yes, this should say "matching or firing" as you suggest.  Matching and firing are usually correlated but not with icount.  You don't want, for instance:
- M-mode-only implementation
- we're in a handler with MIE=0 (which is suppressing icount matching)
- set icount to count a single instruction.  This won't match until after we're out of the handler.
- mret back to non-handler code.  MIE=1 after the mret.
- take an interrupt before executing anything
- the interrupt does an icount match (but not a fire) and sets icount.pending
- we enter the interrupt handler with MIE=0 and pending=1 which (if not suppressed) will cause icount to fire

If icount fires then we'll go to the breakpoint handler and mepc will point to the first instruction of the interrupt handler.  We'll be able to return from the breakpoint handler to the interrupt handler but we lost the address that the interrupt handler should return to.  Firing needs to be suppressed until MIE=1 so we know that the interrupt handler has saved mepc on the stack (or whatever it needs to do).

pdonahue-ventana, 2023-10-10:
After all the PRs, I think that we're done with this one.

closed 2023-10-10 by pdonahue-ventana