riscv/riscv-debug-spec #813: Should tcontrol inhibit icount counting?

scottj97, 2023-03-02:
The spec says that when `tcontrol.mte == 0`:

> Triggers with action=0 do not match/fire while the hart is in M-mode.

And it says about icount:

> When count is greater than 1 and the trigger matches, then count is decremented by 1. (This is true even if the action configured in the trigger is inhibited for some reason.)

The first sentence there suggests that when inhibited by `tcontrol.mte` the count should not decrement, because `tcontrol` inhibits "match/fire" and this uses the same verb "matches".

But the second sentence makes me wonder if `tcontrol` is intended only to inhibit the action configured, and therefore a positive count should still be decremented.

Since the purpose of `tcontrol` is to inhibit reentrancy, it doesn't seem necessary (or desirable?) to inhibit the counting too.

scottj97, 2023-03-02:
Some interesting discussion on this in #555 and #557. IIUC when we added the `pending` bit that issue became moot, so #557 was reverted by #574. But the current text still suggests that counting should be squashed, when that apparently was not the intention after `pending` was added.

timsifive, 2023-03-03:
I don't think we considered icount of M-Mode when a trap handler is being executed.

I'm struggling to come up with an actual use case for it. I guess if you have a debug stub on the target, and it wants to single step, it can use icount for that. So it very carefully crafts code than can set icount to the proper value so that when it returns icount.count=1 and exactly one instruction executes before entering the debug stub handler. Except another exception happens. In that case you'd prefer for icount.count not to be decremented.

Or maybe you're using icount.count to turn trace on and off, in which case... who knows whether you want these trap instructions to be counted or not. If you just want N of your program instructions, then not. If you just want M instructions because that's all that fits in your trace buffer, then you do.

All of this is for some system where you're debugging M-Mode from M-Mode itself, which is clearly asking for trouble and tricky bugs in the debugger. Which is me just wondering if it really matters. Are people going to implement M-Mode only systems and then also icount triggers? Adding U-Mode is probably a lot more useful than icount triggers, and not much more expensive.

timsifive, 2023-03-03:
For consistency, it makes sense to have tcontrol and textra behave the same way.

scottj97, 2023-03-03:
This is confounded by this parenthetical statement in the icount spec:

> When count is greater than 1 and the trigger matches, then count is decremented by 1. (This is true even if the action configured in the trigger is inhibited for some reason.)

Why might the action be inhibited?

In section 5.4 “Native Triggers” it gives two options for preventing reentrancy. The second option is tcontrol. The first option also causes inhibition of breakpoint exceptions sometimes.

Couldn't we say that for consistency, those two mechanisms should behave the same, in terms of blocking the icount decrement?

timsifive, 2023-03-04:
On Fri, Mar 3, 2023 at 11:54 AM Scott Johnson ***@***.***>
wrote:

> This is confounded by this parenthetical statement in the icount spec:
>
> When count is greater than 1 and the trigger matches, then count is
> decremented by 1. (This is true even if the action configured in the
> trigger is inhibited for some reason.)
>
> Why might the action be inhibited?
>
I did some digging, and I'm pretty sure that parenthetical was added in
response to this discussion:
https://github.com/riscv/riscv-debug-spec/pull/665/files/02626239e6f70ec2fa3d0fa767fa7b2638cad336#diff-979319471706782a119ea34ac26af26137b1beedfbeec7603a5a314f841810eb

TLDR:
Q: Does "a mode where the trigger is enabled" just mean the privilege mode
or if action=0 does it also consider the interrupt enable or tcontrol.mte
A: We don't know, but the spec should say *something*.

So we need to revisit that parenthetical, making the spec more clear and
the behavior more consistent.

Tim

> Message ID: ***@***.***>
>

scottj97, 2023-03-28:
> A: We don't know, but the spec should say *something*. So we need to revisit that parenthetical, making the spec more clear and the behavior more consistent.

It would be good to resolve this before the spec is ratified. I'm still not sure what the expected behavior is.

timsifive, 2023-03-30:
The more I look at it, the less sense the parenthetical makes. What does it even mean to inhibit an action?

mte=0 inhibits matching and firing, and #826 makes it so that the same is true for textra. I think I can just remove the parenthetical and call it good. Thoughts?

pdonahue-ventana, 2023-04-01:
Fixed by #829.

closed 2023-04-01 by pdonahue-ventana