riscv/riscv-debug-spec #816: Exceptions in park loop

jrahmeh, 2023-03-06:
For execution based implementations (section A.2), what happens if the code in the "park loop" encounters an exception. Is the exception ignored?  If it is not ignored, does it not destroy the state of the machine (mepc, mcause, mtval ...)?

pdonahue-ventana, 2023-03-07:
A.2 says that the exception is not ignored:
> If an exception is encountered, the hart jumps to an address within the Debug Module.
The code there causes the hart to write to the Debug Module indicating an exception. Then the
hart jumps back to the park loop. The DM infers from the write that there was an exception, and
sets cmderr appropriately.

The Debug Mode section says that exceptions don't update mepc, mcause, mtval, etc.

jrahmeh, 2023-03-08:
It is not clear whether the exception is ignored during the execution of the program buffer code, or in the park loop code, or both.  If would be nice if this is explicitly stated. Something like:  If an exception is encountered by the program buffer code or the park loop code, the hart jumps to an address within the Debug Module.

pdonahue-ventana, 2023-03-29:
Note that if an exception occurs during execution of the park loop then you're probably doing it wrong.  PMP and PMA checks are disabled according to the last paragraph in A.2.  Translation is disabled because it's debug mode (which is like M mode).  Interrupts (including NMI) are disabled in debug mode.  It's a ROM with known contents so there aren't any undefined instructions or ecalls.  The only thing I can think of would be some sort of custom exception, though those should really be disabled in the park loop also.