riscv/riscv-debug-spec #819: [Q] Address field on DMI access for DM result (TDI on JTAG)?

MountaineerGit, 2023-03-07:
Referring to riscv-debug-spec1.0, chapter 6.1.5 Debug Module Interface Access (dmi, at 0x11)

The address-field mentions: "Address used for DMI access. In Update-DR this value is used to access the DM over the DMI."
How about the return-path, answer from DM?
Does the address-field contain the read register-address from the previous transaction or is it implementation defined?

Example communication on JTAG to read data0 register with 7-abits:
```
TDO->0x01000000001  // addr:0x04, data:0x00000000, op:0x1 read; Abstract Data 0 (data0)
TDI<-0x05808000010  // addr:0x16, data:0x02000004, op:0x0 success; Abstract Control and Status (abstractcs)
TDO->0x05800000001  // addr:0x16, data:0x00000000, op:0x1 read; Abstract Control and Status (abstractcs)
TDI<-0x01100025C1C  // addr:0x04, data:0x40009707, op:0x0 success; Abstract Data 0 (data0)    //    is the addr=0x4 here mandatory?
```

timsifive, 2023-03-10:
Looks like this is underspecified.

SiFive cores and spike return the address, as it is in your example.

OpenOCD does not use this return value.

Based on that, a line that explicitly says that the value of dmi.address is unspecified when scanning it out seems reasonable.

Thoughts?

MountaineerGit, 2023-03-13:
It would be great to mention it in the specification!
I'm not sure if other commercial probes like SEGGER or Lauterbach do rely on this address-field to do some kind of data-matching.
I assume the behaviour of SiFive cores is taken as a golden reference. Hence, this address-field could lead to issues on other probes.

timsifive, 2023-03-13:
Is there a benefit to the implementation by allowing dmi.address to be unspecified when read?

I suppose if you implement JTAG with a single shift register that changes size and contents based on IR, and a packet-based bus so that once you send the address you might not keep it around anymore. (As opposed to a simpler bus where the DTM directly drives address onto the bus.) Is that a realistic scenario?

pdonahue-ventana, 2023-03-13:
I'm not sure that I understand the original question.  The table explicitly says that address is UNSPECIFIED after executing a nop or a write and that address is the address that was read from after executing a read.

I seem to recall that this was added about a year ago after some discussion.  Somebody (maybe even me, I don't remember) wanted the flexibility because debuggers shouldn't care (since they ought to be able to remember what they previously wrote).

timsifive, 2023-03-13:
Good point. Thanks for reading more broadly than I did.

closed 2023-03-28 by pdonahue-ventana