riscv/riscv-debug-spec #835: Standardized USB DTM

svobodnik, 2023-04-13:
USB's packet-based nature would be quite amenable to a DTM implementation. Has anyone considered writing a standard for it?

With a USB DTM a mobile/embedded device could provide a USB function which allows external debug access to the SoC (somehow cooperating other functions like MTP, CDC or ADB) in a similar vein to Apple's lighting debug capability. It would be a very powerful debugging tool.

There are some mentions of a possible USB DTM in the spec, but no actual standards or implementation advice. This is just asking for manufacturers to roll their own proprietary DTMs and debug tools, which exactly nobody wants to happen.

svobodnik, 2023-04-13:
Implementing a persistent debugger on a very busy mobile OTG port would be a pain, but that's the SoC designer's job, right?

pdonahue-ventana, 2023-04-13:
There's no chance of having a USB DTM spec before the spec goes to the Architecture Review Committee to be frozen (48 hours from now).  After that, the question is best addressed by the [Debug, Trace, and Performance Monitoring SIG](https://lists.riscv.org/g/sig-debug-trace-perf-mon) as part of its gap analysis.

svobodnik, 2023-04-13:
Edited "standard" to "proposal". I'm not demanding that you immediately push some half-baked USB DTM spec out, I just think this needs attention at some point over the next few years. Didn't mean to imply urgency.

bruceable, 2023-04-13:
The company UltraSoC developed a USB-based debug port:  "UltraSoC pushes
debug over USB" (2015).
https://www.techdesignforums.com/blog/2015/04/22/ultrasoc-usb-debug/

UltraSoC is now part of Siemens, and Iain Robertson is in that division
(see his signature below).  He also chairs the DTPM SIG as Paul Donahue
mentioned
( Debug, Trace, and Performance Monitoring SIG
<https://lists.riscv.org/g/sig-debug-trace-perf-mon> )

Perhaps Iain can comment.

--- Bruce Ableidinger, SiFive, Inc.

[image: image.png]

bruceable, 2023-04-13:
Bruce,
            Not exactly sure what you’re looking for here, but I can confirm that Siemens can provide debug over USB as well as JTAG.  In our architecture the external, off-chip interface/protocol is decoupled from the DTM.  So any existing RISC-V debug module with a JTAG or memory-mapped register interface can communicate with a debugger using JTAG, USB (or even AXI if the debugger is running elsewhere on the same SoC).

Iain

timsifive, 2023-04-13:
I would love it if someone makes a USB proposal, that then can be worked on and eventually incorporated into the Debug Spec. This issue has been raised several times but noone has ever taken the time to write up a proposal to get discussion started.

bruceable, 2023-04-14:
Svobodnik wrote,

"USB's packet-based nature would be quite amenable to a DTM implementation.
Has anyone considered writing a standard for it?

With a USB DTM a mobile/embedded device could provide a USB function which
allows external debug access to the SoC (somehow cooperating other
functions like MTP, CDC or ADB) in a similar vein to Apple's lighting debug
capability. It would be a very powerful debugging tool.

There are some mentions of a possible USB DTM in the spec, but no actual
standards or implementation advice. This is just asking for manufacturers
to roll their own proprietary DTMs and debug tools, which exactly nobody
wants to happen."

Since Siemens/UltraSoC has experience with USB-based debug, I am suggesting
that they consider following Svobodnik's suggestion of writing an RVI
standard for such an interface, or donating such a design for an RVI
standard.  SiFive has donated a number of designs to RVI including Trace
Control for N-Trace and E-Trace, IOMMU, World Guard, and Path Profiler for
the Control Transfer Records task group.

--- Bruce Ableidinger, SiFive, Inc.