riscv/riscv-debug-spec #842: Does icount trigger match on entering Debug Mode due to triggers?

YenHaoChen, 2023-05-03:
The spec says "This (icount) trigger matches on every instruction completed or trap taken ..."
[image]
To my reading, the icount will decrease on ebreak instructions but will not decrease on entering Debug Mode.

I want to clarify the following conditions.
1. Does the icount decrease on firing trigger action=breakpoint when icount.count=1?
2. Does the icount decrease on firing trigger action=debug_mode when icount.count=1?

timsifive, 2023-05-03:
> To my reading, the icount will decrease on ebreak instructions but will not decrease on entering Debug Mode.

Correct. Maybe the spec should be more clear, but in Debug Mode triggers are completely inactive.

icount matches "on every instruction completed or trap taken" and a trap refers to "the transfer of control to a trap handler caused by either an exception or an interrupt" (unpriv spec section 1.6).

> 1. Does the icount decrease on firing trigger action=breakpoint when icount.count=1?

Yes. action=breakpoint explicitly raises "a breakpoint exception" which then results in a trap taken. (I don't think you can mask the breakpoint exception.)

> 2. Does the icount decrease on firing trigger action=debug_mode when icount.count=1?

Only if the instruction is executed. Entering Debug Mode is not a trap, even though in systems with a Debug ROM it looks a lot like it. Debug Spec section A.2:

"When the halt request bit is set, the Debug Module raises a special interrupt to the selected harts. This interrupt causes each hart to enter Debug Mode and jump to a defined memory region that is serviced by the DM and is only accessible to the harts in Debug Mode. When taking this jump, pc is saved to dpc and cause is updated in dcsr. This jump is similar to a trap but it is not architecturally considered a trap, so for instance doesn’t count as a trap for trigger behavior."

YenHaoChen, 2023-05-04:
Another quick question. Does the icount trigger decrease on NMI?

timsifive, 2023-05-04:
I don't think the spec says. My inclination is to say "no" since taking an NMI doesn't execute an instruction (except the ones in the trap handler, which will be counted depending on the modes where icount is enabled).

I'll make a PR and we'll see what people think.

timsifive, 2023-05-04:
BTW, the text you quoted in the very start is from a slightly older version of the spec. The latest (in this git repo) is more clear about icount behavior.

closed 2023-05-04 by YenHaoChen

timsifive, 2023-05-05:
Note that #843 now says the opposite of what I originally told you.

pranjal20498, 2024-06-13:
Does it mean if my count is =5 then my after 5 instructions my count trigger would fire?

rtwfroody, 2024-06-13:
@pranjal20498, yes. That's the simple version of what happens. There are more details in the spec about corner cases etc.