riscv/riscv-debug-spec #844: Use alternative terms for asymmetric communication or control

cetola, 2023-05-08:
Several places in the document reference [communication and control terms](https://github.com/riscv/riscv-debug-spec/blob/cc104d4fea220abbc39d7db83904f0b5cd8aee0a/debug_module.tex#L46) which are not part of our approved list of terms.

Please remove any references to "master" or "slave" and pick a different set of terms. You can see an example of a similar fix from [an IOMMU commit](https://github.com/riscv-non-isa/riscv-iommu/pull/85/commits/ce0151f4ab69eba1148ed68208d660d56911e6a5).

Thanks for your help here.

pdonahue-ventana, 2023-05-08:
In the industry I've seen different people go in different directions (e.g. requester, initiator, controller).  Is there a RISC-V standard term?

cetola, 2023-05-08:
No, rather than create a standard set of terms, we've left it up to the author to chose the text that best fits the situation. Since our specs cover a wide range of applications (extensions, ABIs, etc), we do not want to confine our authors to a given set of terms.

We will be publishing a full list of terms that are not acceptable. That work is an ongoing part of our documentation efforts.

timsifive, 2023-05-09:
Do you have guidance at least? Master and slave are used in different ways probably, but "bus master" and "bus slave" are singular concepts that should all use the same term in RISC-V documentations.

cetola, 2023-05-09:
Personally I liked the solution that AIA used here: https://github.com/riscv/riscv-aia/commit/a990be23969a40cfd86bcd6a38673a15f7ea55f0

I'm open to creating a "best practices" guide, especially if we can thing of some other common terms like "master hart" and bus master".

pdonahue-ventana, 2023-05-24:
I believe that this is completely addressed in #845.

closed 2023-05-24 by pdonahue-ventana

jjscheel, 2023-10-23:
Yes.  I agree.  Thanks for addressing!