riscv/riscv-debug-spec #849: conflicting statements on dscratch registers

eroom1966, 2023-05-18:
There appear to be conflicting statements regarding dscratch registers

```
4.8 Core Debug Registers
The supported Core Debug Registers must be implemented for each hart that can be debugged.
They are CSRs, accessible using the RISC-V csr opcodes and optionally also using abstract debug
commands.
```

the Core debug registers consist of dpc, dcsr, dscratch0, dscratch1

Later in the document
```
4.8.3 Debug Scratch Register 0 (dscratch0, at 0x7b2)
Optional scratch register that can be used by implementations that need it.
```

I think an update is required regarding dscratch registers, either they "**must be implemented**" or they are "**optional**", but cannot be both

JamesKenneyImperas, 2023-05-18:
New parameters `dscratch0_undefined` and `dscratch1_undefined` have been added to allow this to be specified.

timsifive, 2023-05-18:
You're right. Somehow I thought the first quote explicitly mentioned dscratchN. It does not.

I agree with Paul's interpretation. It's a little curious, given how much flexibility people have otherwise, but it doesn't really bother me.

eroom1966, 2023-05-19:
Let me ask the question slightly differently
Could a legal implementation define dpc and dcsr as present, but dscatch0, dscratch1 not present

pdonahue-ventana, 2023-05-19:
> Could a legal implementation define dpc and dcsr as present, but dscatch0, dscratch1 not present

Definitely.  Then dscratch0/1 are not supported registers on this hart.  The requirements in this case are that all harts do not have dscratch0/1 (as required by the first sentence of 4.9 which says that if some other hart supports dscratch0/1 then this hart must do so also) and that hartinfo.nscratch=0 (as required by 3.15.3).

eroom1966, 2023-05-19:
got it - many thanks

closed 2023-05-19 by eroom1966