riscv/riscv-debug-spec #853: Setting both single stepping and icount trigger at the same time

YenHaoChen, 2023-07-06:
Hello,

The spec intends to use the icount trigger as the single step. What will the desired behavior be if a user sets both an icount trigger with action=debug_mode and single stepping?
1. The icount trigger and single stepping behave independently. Thus, the debug mode may enter twice on a single instruction. (The single stepping has a higher priority over the icount trigger if both want to enter debug mode simultaneously.)
2. The single stepping suppresses the icount trigger since there is no reason to enter the debug mode twice. In other words, the icount trigger does not decrease nor fire if dcsr.step=1.

Thank you.

Best regards,
YenHaoChen

timsifive, 2023-07-07:
I don't think there is "desired behavior" for this case, since I can't think of a reason that you'd want to set both at the same time. So the behavior is whatever a literal reading of the spec gives.

Section 4.5.1 says:
> If executing or fetching the instruction causes a trigger to fire with action=1, Debug Mode is re-
> entered immediately after that trigger has fired. In that case cause is set to 2 (trigger) instead of 4
> (single step). Whether the instruction is executed or not depends on the specific configuration of
> the trigger.

That would imply that the trigger supersedes the single step. Debug Mode is entered exactly once for that instruction.

YenHaoChen, 2023-07-09:
Thank you for the reply.

The spec clarifies later, "... if a single instruction is stepped with both icount and step then the step has priority."
[image]

From my reading, the mcontrol/mcontrol6 with timing=before (cause=2) supersedes the single step, but the icount, etrigger/itrigger, and mcontrol/mcontrol6 with timing=after have a lower priority to the single step (cause=4). Is this correct? Do both cases enter Debug Mode only once for an instruction?

If the single step supersedes icount, and the Debug Mode only enters once, will the icount still decrease even if it will not have an effect with dcsr.step=1?

timsifive, 2023-07-10:
On Sat, Jul 8, 2023 at 9:43 PM YenHaoChen ***@***.***> wrote:

> The spec clarifies later, "... if a single instruction is stepped with
> both icount and step then the step has priority."
>
> [image: image]
> <https://user-images.githubusercontent.com/39526191/252125735-d6ee882a-847b-453c-b591-d561ca29288d.png>
>
Table 4.2 shows what DCSR reflects, when multiple causes occur at the same
time. It does not imply anything about the relative priority of those
causes during execution.
> From my reading, the mcontrol/mcontrol6 with timing=before (cause=2)
> supersedes the single step, but the icount, etrigger/itrigger, and
> mcontrol/mcontrol6 with timing=after have a lower priority to [...]
Section 4.5.1 says:

> If executing or fetching the instruction causes a trigger to fire with
> action=1, Debug Mode is re-
> entered immediately after that trigger has fired. In that case cause is
> set to 2 (trigger) instead [...]

I wrote that 6 years ago. I think it says that triggers, regardless of type
or timing, have higher priority than dcsr.step. I'm OK with that (although
I may want to clarify that it only applies to precise triggers).

Is that behavior reasonable? Is there an obvious clarification that is
necessary?

Tim

> Message ID: ***@***.***>
>

YenHaoChen, 2023-07-11:
Thank you. I'm OK with that.

When reading Section 4.5.1, I believe all triggers have higher priority than dcsr.step. Table 4.2 also lets the priority of the trigger's cause value be higher than the step's cause value.

However, I am confused when reading the commentary "... if a single instruction is stepped with both icount and step then the step has priority." From the above discussion, the icount trigger has a higher priority than dcsr.step (and suppressing it).

I do not understand the commentary. Can you share some words about the commentary?

timsifive, 2023-07-12:
That is conflicting and confusing.

I'm struggling with finding the right answer.

Triggers with timing=before obviously take precedence over dscr.step.
If timing=after, the trigger may not fire until many instructions have been
executed. There is the special case (new mcontrol6 only) where we can
differentiate between "after" and "immediately after," but it would keep
the spec a lot simpler to treat those the same. So for some hardware, it's
simply not possible to have timing=after take precedence over dcsr.step.

Which is the opposite of what I told you the other day. This would make the
commentary correct, and the text in 4.5.1 to be incorrect.

What are your thoughts? @pdonahue-ventana, what do you think?

YenHaoChen, 2023-07-13:
@timsifive's proposal makes more sense to me, although the text in 4.5.1 may need to change the object "all triggers" to "triggers with timing=before."

Additionally, adding an entry of trigger with timing=after at the bottom of Table 4.2 may present the proposal more transparent.
```
| cause encoding | Cause                                                 |
|        5       | resthaltreg                                           |
|        6       | halt group                                            |
|        3       | haltreq                                               |
|        2       | trigger with timing=before                            |
|        1       | ebreak                                                |
|        4       | step                                                  |
|        2       | trigger with timing=after, icount, itrigger, etrigger |
```

Oil-Bao, 2025-10-21:
Hi,
I also have the same question.
when in a M+S+U hart, dcsr.step = 1, etrigger.action=0, and the tdata2[2]=1

> ( to trap on an illegal instruction, the debugger sets bit 2 in tdata2. )

in following  scenario(hart in S-mode):
hart status: in S-mode, medeleg[2] = 1. dcsr.step = 1. and medeleg[3]=0
so, when hart execute instruction, and it's illegal instruction, so this trap would delegate to S-mode, scause = 2. then enter in debug-mode(cause dcsr.step=1). so the etrigger is losed. 

the 5.4 chapter,native triggers has:

> The hardware prevents triggers with action=0 from matching or firing while in M-mode and while
MIE in mstatus is 0. If medeleg [3]=1 then it prevents triggers with action=0 from matching or
firing while in S-mode and while SIE in sstatus is 0. .....

and now the etrigger.action=0, and  medeleg[3]=0, mstatus.MIE=1(cause illegal trap delegate), so it should not be losed(etrigger)

so the single-step cross timing-after trigger(etrigger/itrigger) action= 0, how to do this?

I have looked some issues discussed similar problems, but no golden-answer mentioned when this happened.
#890 https://lists.riscv.org/g/tech-debug/topic/108301145#msg1770
@pdonahue-ventana @timsifive 

Thank you.

Best regards,
Oil